dan-v/rattlesnakeos-stack #3: does not build

cryptocub3, 2018-06-21:
I am using latest version.

Tried 2 times, the ec2 instances spawned but looking at the cpu usage was activity only about 30 minutes after I think the build process quits with an error, but I don't have experience with aws and I can't look in logs if you can guide me I will leave the instance running more time to give you some debugging feedback.

dan-v, 2018-06-21:
Hi, please make sure to use the provided issue template in the future. Knowing what platform you are on for example would be helpful.

The build process can take 4+ hours on initial build. Looking at the CPU usage is not a great mechanism for determining build status. Here are some debugging tips:
* Make sure you setup email notifications: https://github.com/dan-v/rattlesnakeos-stack#first-time-setup-after-deployment. This should notify you when a build starts and when it stops either due to success or failure. If you aren't getting a Success or Error email then it's still building.
* If you are getting an error email then let's look at the logs. On build failure/success, the instance should terminate and upload its logs to S3 bucket called `<stackname>-logs` and it's in a file called `<device>/<timestamp>`. If you can provide the log file that would be useful in debugging. Note this log file can be quite large.
* Depending on your comfort level It's also possible to watch the live status of the build by SSHing into the instance. In order to be able to do that you need to:
  * Create an SSH keypair in EC2 (https://us-west-2.console.aws.amazon.com/ec2/v2/home?region=us-west-2#KeyPairs:sort=keyName) and download it
  * Pass an additional flag to rattlesnakeos-stack command (--ssh-key yourkeypairname)
  * Kick off a manual build through AWS Lambda console (see FAQ for details on that)
  * You should be able to SSH into the instance (ssh -i yourkeypairname.pem ubuntu@yourinstancepublicip): 'tail -f /var/log/cloud-init-output.log' to see progress

dan-v, 2018-06-21:
Gonna close this one in favor of issue #4 . Likely the same issue.

closed 2018-06-21 by dan-v