dan-v/rattlesnakeos-stack #4: Build failed

tectiv3, 2018-06-21:
### Prerequisites

* [x] Are you running the latest version?

### Description

Build failed after about 30 minutes, email text:

`RattlesnakeOS build FAILED for sailfish (date=2018.06.21.09 aosp_build=OPM4.171019.021.D1 aosp_branch=android-8.1.0_r31 kernel_branch=android-msm-marlin-3.18-oreo-m4)`

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** build succeed

**Actual behavior:** build failed

### Version

You can get this information from executing `rattlesnakeos-stack --version`.

```
 ./rattlesnakeos-stack --version
Error: unknown flag: --version
```

### Platform
`macOS`

### Full Command Used for Setup

```
 ./rattlesnakeos-stack --region ap-northeast-1 --name rattlesnakeos-tectiv3 --device sailfish
```

### Log output

Parts with `error` in logs:

```
Warning: File /home/ubuntu/.android/repositories.cfg could not be loaded.
[=                                      ] 4% Fetch remote repository...         
[==                                     ] 5% Fetch remote repository...         
[==                                     ] 6% Fetch remote repository...         
[==                                     ] 6% Computing updates...               
[===                                    ] 8% Computing updates...               
[===                                    ] 8% Computing updates...               
[===                                    ] 10% Computing updates...              
[=======================================] 100% Computing updates...             
download failed: s3://rattlesnakeos-tectiv3-release/chromium/revision to - An error occurred (404) when calling the HeadObject operation: Not Found
Chromium current: 
Chromium latest: 67.0.3396.87
Building chromium 67.0.3396.87
~/rattlesnake-os ~/chromium /
Cloning into '/home/ubuntu/depot_tools'...
```

```

src/v8 (ERROR)
```

```
Error: 138> 
138> ____ src/v8 at a64db1ffcc1f9521a4e338b539607415fa64131e
138> 	You have unstaged changes.
138> 	Please commit, stash, or reset.
```

tectiv3, 2018-06-21:
I will try to update build script to retrieve chromium source via `https://commondatastorage.googleapis.com/chromium-browser-official/chromium-67.0.3396.87.tar.xz`

tectiv3, 2018-06-21:
that was a bad idea... Source downloaded that way is left in half-assed state. No buildtools and I couldn't find a way to get them.

tectiv3, 2018-06-21:
```
~/chromium/src/v8$ git status
HEAD detached at 6.9.192
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   tools/link_clicker.extension/icon.png
	modified:   tools/turbolizer/hide-selected.png
	modified:   tools/turbolizer/hide-unselected.png
	modified:   tools/turbolizer/layout-icon.png
	modified:   tools/turbolizer/left-arrow.png
	modified:   tools/turbolizer/live.png
	modified:   tools/turbolizer/right-arrow.png
	modified:   tools/turbolizer/search.png
	modified:   tools/turbolizer/search2.png
	modified:   tools/turbolizer/turbolizer.png
	modified:   tools/turbolizer/types.png
	modified:   tools/turbolizer/upload-icon.png
```

something modifies those files and that makes `gclient sync` fail.

dan-v, 2018-06-21:
Thanks for the detailed report @tectiv3 ! I am not sure why chromium build is failing, but I will try to investigate. I have built the same version successfully with this setup and once it is built successfully it is uploaded to S3 and re-used until a new stable version of Chromium is released.  I'll delete my Chromium build and see if I can reproduce your issue.

Until then you could upload a pre-built chromium to your S3 bucket to get past that step. I've temporarily uploaded the requires files to S3. You can:
* In the S3 bucket (rattlesnakeos-tectiv3-release) create a folder 'chromium'
* Put the following two files in there: 
  * https://s3-us-west-2.amazonaws.com/temp-chromium-build/MonochromePublic.apk
  * https://s3-us-west-2.amazonaws.com/temp-chromium-build/revision
* Kick off another build manually

dan-v, 2018-06-21:
I was able to reproduce the issue and will try to figure out a workaround.

dan-v, 2018-06-21:
I've committed a fix that gets me past that issue, although it's still building (takes a while) so not 100% verified. You can grab the new release here: https://github.com/dan-v/rattlesnakeos-stack/releases/tag/v0.0.3. I'll keep this issue open until the build finished and I update my phone.

tectiv3, 2018-06-21:
Thank you! I was able to get past this, but now I'm getting:
```
Your identity is: unknown <unknown@localhost>
If you want to change this, please re-run 'repo init' with --config-name

repo has been initialized in /home/ubuntu/rattlesnake-os
sed: can't read build/make/target/product/core_minimal.mk: No such file or directory
```

dan-v, 2018-06-21:
@tectiv3  - cool! I'll take a look once my build gets to that point. I think it know the issue on this one. I have a pre-built AMI that has all of these sources already synced so I can work on this quicker. I'm guessing that file doesn't exist until after the repo sync on a clean build. I'll test out that theory and get you a new build.

tectiv3, 2018-06-21:
Thank you :) Also this error happens almost immediately if chromium is already in s3.

dan-v, 2018-06-21:
Alright here's a fix: https://github.com/dan-v/rattlesnakeos-stack/releases/tag/v0.0.4. Let's see what the next issue is :)

tectiv3, 2018-06-22:
got
```
HTTP request sent, awaiting response... 403 Forbidden
2018-06-22 04:00:36 ERROR 403: Forbidden.
```
and
```
upload: release-sailfish-2018.06.22.02/sailfish-ota_update-2018.06.22.02.zip to s3://rattlesnakeos-tectiv3-release/sailfish-ota_update-2018.06.22.02.zip
fatal error: An error occurred (404) when calling the HeadObject operation: Key "sailfish-ota_update-.zip" does not exist
```
But build succeed

dan-v, 2018-06-22:
I think this happens because you don't have any existing OTA updates - it's trying to cleanup old OTA updates, and it fails to get a proper `old_date` variable. It's currently set to ignore the error, so hopefully didn't impact the build. Would be nice to cleanup though.

dan-v, 2018-06-22:
Oh, also hooray for successful build! Let me know if you flash it successfully - can add official support for sailfish to the list.

tectiv3, 2018-06-22:
Everything is fine!

closed 2018-06-22 by tectiv3

dan-v, 2018-06-22:
awesome!

dan-v, 2018-06-22:
In those instructions I didn't see a final step to relock your bootloader, which is important if you care about the verified boot portion of this. See details here: https://copperhead.co/android/docs/install#flashing-factory-images. Do note that locking your bootloader will wipe the data again.

tectiv3, 2018-06-22:
wow, thank you!

tectiv3, 2018-06-24:
sorry to bother you again, after I locked bootloader phone is now showing warning on reboot that it has loaded different OS and shows some fingerprint. is it ok?

captainhook, 2018-06-24:
This is very normal

Kindest Regards,
Abdullah-Fahim Miah

Disclaimer: This e-mail is intended solely for the person(s) to whom it is addressed and may contain confidential or privileged information. If you are not the addressee, dissemination, copying or use of this e-mail or any of its content is prohibited and may be unlawful. If you are not the intended recipient please inform the sender immediately and destroy the e-mail, any attachment(s) and any copies.

dan-v, 2018-06-24:
Indeed this is expected. Basically just warning you that you don't have a Google signed OS.

tectiv3, 2018-06-24:
got it, thanks!

sjdrc, 2018-06-25:
I'm getting failed builds due to chromium with latest master
```
ninja: error: '../../third_party/libsync/src/sync.c', needed by 'obj/third_party/libsync/libsync/sync.o', missing and no known rule to make it
```
Any ideas on how to fix this?

dan-v, 2018-06-25:
@sjdrc - please open a new issue with all of the requested details of the issue template.