dan-v/rattlesnakeos-stack #7: Building chromium fails

sjdrc, 2018-06-25:
### Prerequisites

built from master

### Description

Build fails with the following error during the build of chromium
```
ninja: Entering directory `out/Default/'
ninja: error: '../../third_party/libsync/src/sync.c', needed by 'obj/third_party/libsync/libsync/sync.o', missing and no known rule to make it
```
### Steps to Reproduce

Compile the go application and run it to provision AWS services

**Expected behavior:** 

**Actual behavior:** [What actually happened]

### Platform
Linux

### Full Command Used for Setup
./rattlesnakeos-stack --region ap-southeast-2 --name rattlesnakeos-sjdrc --device taimen --ssh-key rattlesnakeos-sjdrc --spot-price 1.2 --prevent-shutdown

### Email Notification output
Should look something like this:
```
Received no error notification

### Log output

[1529949837.txt](https://github.com/dan-v/rattlesnakeos-stack/files/2134249/1529949837.txt)

dan-v, 2018-06-25:
@sjdrc  - where are you seeing that error? I don't see it in the provided log. I see the following which looks like the process was killed:
```
error: pathspec '67.0.3396.87' did not match any file(s) known to git.
Removing chrome/android/profiles/chrome-profile-3309-text.prof
Removing third_party/android_build_tools/aapt2/lib64/
Removing third_party/shaderc/
Killed
Cloud-init v. 17.1 running 'modules:final' at Mon, 25 Jun 2018 17:30:22 +0000. Up 10.06 seconds.
```

Also, can you try to grab latest master (there was a release today). There is a fix in there that removed a flag `--no-history` from the fetch (with this flag it isn't able to check out specific release of chromium): https://github.com/dan-v/rattlesnakeos-stack/commit/a4ac7606a6cb437adcf7a9b8184ad7c37e568fbe#diff-b868d76e2ffe8cd1568e4cc0d33fec29L236.

sjdrc, 2018-06-26:
Whoops, I re-ran the build again over ssh so that's why logs might have not been generated.

dan-v, 2018-06-26:
@sjdrc - let me know if running latest solves your problem.

sjdrc, 2018-06-26:
It looks like chromium, fdroid, and fdroid-priv managed to build but the instance is still running and there is no recent log in the logs bucket. I just logged on over ssh and CPU usage is still huge so I guess that the build is still running.

dan-v, 2018-06-26:
If you are logged in over SSH - you can run 'tail -f /var/log/cloud-init-output.log' to see progress. The logs don't upload to S3 until it's failed or completed successfully.

sjdrc, 2018-06-26:
Looks like the build worked. Thanks for your help.

dan-v, 2018-06-26:
Let me know if you flash this on your device successfully. If so, I will update the project to say official support for Pixel 2 XL.

sjdrc, 2018-06-26:
Flashed device, touchscreen not working at all. Attempted a new build and that failed. I'll make a new thread because this particular issue with building chromium is resolved.

closed 2018-06-26 by sjdrc