dan-v/rattlesnakeos-stack #8: RattlesnakeOS problems

tectiv3, 2018-06-26:
### Prerequisites

* [x] Are you running the latest version?

### Description

I have a few problems with this ROM.
1. Defunct search bar on home screen. It doesn't work and couldn't be removed.
2. Alarms don't fire, unless phone is unlocked and even then `clock is stopped` displayed after alarm goes on.
3. Very poor camera quality. Not a bug per se, but [camera nx](https://chromloop.com/2018/03/apk-camera-nx-7-4-base-google-camera-5-2-features-bring-back-hdr-nexus2015/) doesn't work on this ROM for some reason (just crashing).

That's it so far. Do you also have those issues or they are specific for sailfish?

dan-v, 2018-06-26:
1. Can be added back by included QuickSearchbar back , but it also leaves an user facing app around that I didn't want. I'm open to adding back, just didn't want to see the extra app laying around. I use a different launcher, so I don't have the defunct search bar.
2. I also have this problem with the alarm clock, I captured a stack trace for it with logcat but I haven't been able to find a resolution to it yet. As a workaround, I'm currently using an alarm app installed through Yalp store called 'Alarm' (com.dehox.alarm) - that has been working fine. Definitely want to track this one down though.
3. I don't think stock AOSP has any of the support for fancy camera features that Google provides. You can take a look at: https://copperhead.co/android/docs/usage_guide#advanced-camera-features for what was recommended with CopperheadOS.

dan-v, 2018-06-26:
Ha, better than setting everything up on CopperheadOS and having to re-install I guess :)

dan-v, 2018-06-26:
I think i tracked down the alarm clock issue (https://github.com/AndroidHardeningArchive/platform_packages_apps_DeskClock/commit/d745babfba6ece923db55dbef14f8c32b2f86d0e). It appears to be an upstream issue that still isn't fixed. I have a fix on dev branch that I'm testing and will push to master if it fixes the alarm clock issue.

tectiv3, 2018-06-26:
Any luck?

dan-v, 2018-06-26:
Alright - fix for alarm clock is working for me and merged: https://github.com/dan-v/rattlesnakeos-stack/releases/tag/v0.0.10

* For item 1, I think I'd prefer to have a defuct bar there rather than making changes to Launcher3 as as that's one more thing to maintain. You can see CopperheadOS changes removing it here: https://github.com/AndroidHardeningArchive/platform_packages_apps_Launcher3/commits/oreo-m2-s3-release
* For item 3, I played around with the camera a bit and it appears to be the same as on CopperheadOS, so not sure what I can fix there at the moment.

tectiv3, 2018-06-26:
Thank you, I will go ahead and build it!
1. yeah, I switched to lawncher, so it's no longer a problem.
3. is there any way to know why an app crashes on launch?

dan-v, 2018-06-26:
cool! having a working alarm clock is nice 👍if you want to try to debug why an app is crashing you can:
* on computer, install adb
* on phone, enable developer options and usb debugging
* on computer run 'adb logcat' and then launch the app that is having issues to see stack trace

tectiv3, 2018-06-26:
Thanks!
I had to use `force`. It will rebuild everything and create an ota file?

dan-v, 2018-06-26:
Ah yes, I was going to mention that. That setting will just force a new build even though all of the versions are the same as last build. It won't force chromium rebuild though - as that's pretty expensive. After you kick off a build, you might want to run rattlesnakeos-stack without the --force flag just so you don't run any builds you don't need to.

tectiv3, 2018-06-26:
ok, camera crashes with
`
java.lang.SecurityException: Failed to find provider com.google.android.gsf.gservices for user 0; expected to find a valid ContentProvider for this authority
`
I suppose it requires GP services for some reason? Ads may be...
microG is out of the question since they require signature spoofing and it's just crazy.

dan-v, 2018-06-26:
are you trying to run Google camera? it doesn't work without google services..

tectiv3, 2018-06-26:
yeah:)

cryptocub3, 2018-06-26:
can confirm alarm is working.

some features I would personally like:
- custom boot animation 
- scrambled unlock pin numbers ( like CHOS used to have )
- wireguard I can see is work in progress ( really nice )
- dnscrypt ( or at least force custom dns servers ) 
- a tool like https://f-droid.org/en/packages/streetwalrus.usbmountr/ would be nice to integrate then we could use the phone to boot custom OS ( don't think achievable without su presence )

thank you for your work.

dan-v, 2018-06-26:
Hey @cryptocub3 ! I'm glad to hear you got it up and running! I'm trying to keep the number of modifications to AOSP to a minimum so that there is less maintenance work involved. I'm hoping some of the additional security/privacy features in Android P will fill the void here when it is released. My thoughts on your bullet points:
* I think a custom boot animation would be nice 
* Not likely to add scrambling pin unlock as I don't want to maintain this
* tectiv3 is working on wireguard kernel support - if he can get it working it will be an optional feature that you have to enable with flag. the other option that works now is using userland version of wireguard.
* Not likely to add dnscrypt or force custom DNS servers. I don't think CopperheadOS did this either and basically only updated the fallback DNS servers which are only used if you aren't provided a DNS server from DHCP. If you have the latest version of Chromium built you can enable DNS over TLS at least for your browser right now (https://github.com/bromite/bromite/wiki/Enabling-DNS-over-HTTPS). Android P will have system level support for DNS over TLS, but not sure if setting this up will override DNS servers when on mobile network (hopefully).  
* USB Mountr looks neat, but not something I want to bake into the OS.

tectiv3, 2018-06-27:
Using wireguard as a kernel module isn't possible without `su` for some reason. Userland implementation works good though, using it in always on mode plus kill switch.
But it is the major battery drainer according to stats.

dan-v, 2018-06-28:
Alright gonna close this one out.

closed 2018-06-28 by dan-v