dan-v/rattlesnakeos-stack #9: Key error for device not being built

sjdrc, 2018-06-26:
### Prerequisites
Running latest master. Building for taimen (Pixel 2 XL)

### Description
Build fails with the error
```ln: failed to create symbolic link '/home/ubuntu/rattlesnake-os/kernel/google/marlin/verity_user.der.x509': No such file or directory```

### Steps to Reproduce
1. Build latest master
2. Run full command
3. Trigger build from lambda

### Platform
Linux

### Full Command Used for Setup
```./rattlesnakeos-stack -r ap-southeast-2 -n rattlesnakeos-sjdrc -d taimen --ssh-key rattlesnakeos-sjdrc --spot-price 1.2```
### Email Notification output
Should look something like this:
```
  Device: taimen
  Stack Version:  WARNING: you should upgrade to the latest version: 0.0.10
  Build Date: 2018.06.26.11
  Elapsed Time: 4hrs 46min 42sec
```
(I've compiled from master)

### Log output
[1530031482.txt](https://github.com/dan-v/rattlesnakeos-stack/files/2139247/1530031482.txt)

sjdrc, 2018-06-26:
Tried latest release, getting this issue
```
$ ./rattlesnakeos-stack -r ap-southeast-2 -n rattlesnakeos-sjdrc -d taimen --ssh-key rattlesnakeos-sjdrc --spot-price 1.2 --prevent-shutdown
INFO[0000] Checking AWS credentials                     
INFO[0000] Creating S3 bucket rattlesnakeos-sjdrc       
INFO[0000] Downloading Terraform binary from URL: https://releases.hashicorp.com/terraform/0.11.7/terraform_0.11.7_linux_amd64.zip 
INFO[0001] Creating AWS resources                       
panic: runtime error: invalid memory address or nil pointer dereference
	panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x831212]

goroutine 1 [running]:
github.com/dan-v/rattlesnakeos-stack/stack.(*TerraformClient).Cleanup(0x0, 0xe, 0x7ffdfc557d04)
	/go/src/github.com/dan-v/rattlesnakeos-stack/stack/terraform_client.go:112 +0x22
panic(0x8f0720, 0xcd1fa0)
	/usr/local/go/src/runtime/panic.go:502 +0x229
github.com/dan-v/rattlesnakeos-stack/stack.(*TerraformClient).Apply(0x0, 0x1, 0x1)
	/go/src/github.com/dan-v/rattlesnakeos-stack/stack/terraform_client.go:89 +0x56
github.com/dan-v/rattlesnakeos-stack/stack.AWSApply(0x7ffdfc557ced, 0x13, 0x7ffdfc557cdb, 0xe, 0x7ffdfc557d04, 0x6, 0x9875dd, 0xc, 0x7ffdfc557d36, 0x3, ...)
	/go/src/github.com/dan-v/rattlesnakeos-stack/stack/aws.go:74 +0x16d
main.glob..func2(0xcd94e0, 0xc42018f130, 0x0, 0xb)
	/go/src/github.com/dan-v/rattlesnakeos-stack/main.go:28 +0x223
github.com/dan-v/rattlesnakeos-stack/vendor/github.com/spf13/cobra.(*Command).execute(0xcd94e0, 0xc42001e0d0, 0xb, 0xb, 0xcd94e0, 0xc42001e0d0)
	/go/src/github.com/dan-v/rattlesnakeos-stack/vendor/github.com/spf13/cobra/command.go:760 +0x2c1
github.com/dan-v/rattlesnakeos-stack/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0xcd94e0, 0x0, 0x99d034, 0x57)
	/go/src/github.com/dan-v/rattlesnakeos-stack/vendor/github.com/spf13/cobra/command.go:846 +0x30a
github.com/dan-v/rattlesnakeos-stack/vendor/github.com/spf13/cobra.(*Command).Execute(0xcd94e0, 0xc4201c3f78, 0xc420088058)
	/go/src/github.com/dan-v/rattlesnakeos-stack/vendor/github.com/spf13/cobra/command.go:794 +0x2b
main.main()
	/go/src/github.com/dan-v/rattlesnakeos-stack/main.go:72 +0x2d
```

I was able to wget the terraform zip, so I'm not sure what's going wrong.

dan-v, 2018-06-26:
Thanks for the detailed report @sjdrc ! The first issue I need to do some digging. Pixel 2 doesn't use verity keys, so need to figure out what the right procedure is for kernel build with verified boot. Is the second issue reproducible on your end?

sjdrc, 2018-06-26:
I deleted absolutely everything and built again with 0.0.10 and it provisioned all the resources correctly.

dan-v, 2018-06-26:
ok - very strange. i'll look into first issue.

dan-v, 2018-06-26:
ok i've pushed a new release: https://github.com/dan-v/rattlesnakeos-stack/releases/tag/v0.0.11. you can just pull latest from master and build also.  next build attempt should be quite a bit quicker as you shouldn't need to compile chromium. thanks for testing this out.

sjdrc, 2018-06-27:
```
/home/ubuntu/build.sh: line 427: syntax error near unexpected token `)'
/home/ubuntu/build.sh: line 427: `    sed -i 's/'${OFFICIAL_FDROID_KEY}'")/'${unofficial_marlin_releasekey_hash}'"),\n            new Pair<>("org.fdroid.fdroid", "'${unofficial_marlin_platform_hash}'")/' \'
```

Got this error from running latest release

dan-v, 2018-06-27:
Alright will take a look at this tonight. Thanks for hanging in there!

dan-v, 2018-06-27:
Alright pushed new release: https://github.com/dan-v/rattlesnakeos-stack/releases/tag/v0.0.12. We'll see what breaks next. I kicked off my own build w/ taimen this time - so maybe i'll find an issue before you.

dan-v, 2018-06-27:
Looks like it built successfully - the real question which I can't test is if it flashes successfully.
```
RattlesnakeOS Build SUCCESS
  Device: taimen
  Stack Version: 0.0.12
  Build Date: 2018.06.27.02
  Elapsed Time: 2hrs 40min 9sec
  AOSP Build: OPM4.171019.021.E1
  AOSP Branch: android-8.1.0_r32
  Kernel Branch: android-msm-wahoo-4.4-oreo-m4
  Chromium Version: 67.0.3396.95
  F-Droid Version: 1.2.2
  F-Droid Priv Extension Version: 0.2.8
```

dan-v, 2018-06-27:
Pushed a new version that has the expected values for Pixel 2 and Pixel 2 XL: https://github.com/dan-v/rattlesnakeos-stack/releases/tag/v0.0.13

```
RattlesnakeOS Build STARTED
  Device: taimen
  Stack Version: 0.0.13
  Build Date: 2018.06.27.16
  Elapsed Time: 0hrs 0min 7sec
  AOSP Build: OPM2.171026.006.C1
  AOSP Branch: android-8.1.0_r30
  Kernel Branch: android-msm-wahoo-4.4-oreo-m2
  Chromium Version: 67.0.3396.95
  F-Droid Version: 1.2.2
  F-Droid Priv Extension Version: 0.2.8
```

sjdrc, 2018-06-28:
Thanks, I'm running a new build now. Will report once I've flashed my phone.

dan-v, 2018-06-28:
@sjdrc - let me know how it goes. I have another set of changes that I'd love to have you test with taimen build. The current build you are running builds the kernel unnecessarily. The latest code I just pushed to master skips building the kernel for Pixel 2 and Pixel 2 XL and I'd love to see if that works as expected.

sjdrc, 2018-06-28:
The EC2 instance seemed to be going fine until it terminated randomly leaving no logs and no completed build, but chromium, fdroid, and fdroid-dev were all compiled

sjdrc, 2018-06-28:
Bit strange that it died without warning, because I had --prevent-shutdown set

dan-v, 2018-06-28:
Yeah, this can unfortunately happen with spot instances (--prevent shutdown won't stop that). If the price exceeds your max bid it will shutdown. You can either wait for a time when price isn't as high or crank up the max bid price to avoid shutdown of your instance.

dan-v, 2018-06-28:
If you look at the terminated instance in the console it should have a field called 'State transition reason message' - and if it says not user terminated then that is what happened.

sjdrc, 2018-06-28:
Oh damn! I had it at 1.20 up from default thinking that should be enough :(

sjdrc, 2018-06-28:
The on demand price graph shows it as 1.04 for the last 3 hours

dan-v, 2018-06-28:
yeah - it unfortunately all depends on demand for those instance. i previously had a few different instance types (rather than just c4.4xlarge) and it would choose the cheapest at build time. maybe i'll think about adding that back in. i only took it out because the c5.4xlarge was running into kernel panics during build for some reason.

sjdrc, 2018-06-28:
Anyway, I'm ready to run another build. What changes would you like me to test?

dan-v, 2018-06-28:
Cool. If you pull latest master - give that a shot. Should be quicker also as there is no kernel build for Pixel 2 and Pixel 2 XL.

dan-v, 2018-06-29:
any luck @sjdrc ?

sjdrc, 2018-07-01:
Hey, sorry for the delay, I've been swamped with work for the last few days. The last build also failed, but strangely the logs say that AOSP build is up to date.
[latest.log](https://github.com/dan-v/rattlesnakeos-stack/files/2152892/latest.log)

dan-v, 2018-07-01:
No worries @sjdrc . It looks like everything has checkpointed at latest in your stack (I think some of the code checkpoints a little too early). You'll need to run `--force`. Also note, there is an upstream build issue with latest Chromium that I've committed a workaround to master for.

sjdrc, 2018-07-02:
It seems that my instance keeps getting killed by spot pricing changing, even though I have my limit set to $2. Do you think it would be possible to make the EBS volume persist so that building images will be much quicker, not having to check out the source and do a full build every time?

dan-v, 2018-07-02:
@sjdrc - it must be the region that you are doing builds in. In us-west-2 where I am running builds, i see the price around $.30 on average. It would be possible to persist the EBS volume, but when I calculated the pricing it wasn't worth it ($20 a month for 200GB volume). 

Some possible options for now:
* You could do builds in another region with cheaper spot instances
* It is possible to create your own AMI and pass that as a flag to rattlesnakeos-stack tool. You could for example, spin up an ubuntu 16.04 image, download all of chromium and AOSP to appropriate locations and then save that as a new AMI. I haven't tested this in a while though and it may break the script as it may try to download something that already exists and fail.

Some possible future options:
* Could make instance type a user facing option and you could pick a cheaper instance type for your region
* Could make it possible to choose between spot instance or on demand instance. On demand instances of c4.4xlarge in ap-southeast-2 are $.90 per hour, so may be cheaper in that case.

sjdrc, 2018-07-02:
Well, the build finally finished! I'm going to flash and test my phone now.

dan-v, 2018-07-02:
Awesome! Hoping that it works!

sjdrc, 2018-07-02:
Great success! Calling works! Thanks for your help through all this!

Do you have a list of features you're wanting to implement? I'm not experienced with go but I'm going to be mucking around with your scripts anyway and I'd like to contribute if something I build helps.

dan-v, 2018-07-02:
That's awesome! Super glad it worked out. I'd love some contributions to the project. I'm pretty content with the base image at this point - not looking to add any additional software or anything as I'm trying to keep this as lean as possible. But there is a lot of work that could be done to make it a more reliable/better build process (as witnessed with the issues you ran into). Right now it's still in fairly rough shape as I just took a shell script and threw it inside go - which was the quickest way to get up and running. In the long run, I'd love to have it run the build process in a real programming language rather than shell script, whether that be go, python, etc. But for short term, I think making incremental improvements on the shell script is best option. I'll work on jotting some improvement ideas as github issues.

closed 2018-07-02 by dan-v