dan-v/rattlesnakeos-stack #11: Privileged permissions not granted to fdroid extension

sjdrc, 2018-07-03:
### Prerequisites

Running version 0.0.14

### Description

Unable to install apps from FDroid due to ungranted permissions to the extension

### Steps to Reproduce

1. Flash Pixel 2 XL with build from latest version of stack.
2. Attempt to install an app from FDroid
3. Error message appears

### Platform
Linux

### Full Command Used for Setup
From previous issue. I'm on a train and will update when I get to the office.

### Email Notification output
N/A but will attach later if required

### Log output
N/A but will attach later if required

dan-v, 2018-07-03:
I think i know what issue is here - I'll post a fix in just a bit.

sjdrc, 2018-07-05:
Package installation from F-Droid works now!

dan-v, 2018-07-05:
Awesome! Let me know if you bump into anything else.

closed 2018-07-05 by dan-v

xCapRx, 2019-01-17:
I have the same problem and I'm using an SM-G900V S5. I tried clearing the memory and cache for F-Droid and granting root permissions to the extension through ISU but it still didn't work.