dan-v/rattlesnakeos-stack #13: New AWS account does not allow SSH access

bradyschulman, 2018-07-06:
### Prerequisites

Yes

### Description

When rattlesnakeos-stack provisions to a new AWS account, it does not add tcp/22 access to the default security group.

### Steps to Reproduce

TCP/22 should be allowed by default in the default security group or it should be added to the README that this is a required step for ssh troubleshooting.

### Platform
Linux

### Full Command Used for Setup
./rattlesnakeos-stack --region us-east-2 --name ros-mine --device walleye --force --ssh-key us-east-2 --force

### Email Notification output
Should look something like this:
```
None
```

### Log output
None.

dan-v, 2018-07-06:
The default security group should have SSH open by default unless you modified this. I can add a note in README to double check though.

dan-v, 2018-07-06:
You were correct, it appears default does not have open SSH access. I've updated README.

closed 2018-07-06 by dan-v