dan-v/rattlesnakeos-stack #15: Chromium build error for 68.0.3440.50 & walleye

chuckb7, 2018-07-06:
### Prerequisites

Yes. 0.15

### Description
ninja: Entering directory `out/Default/'
ninja: error: '../../third_party/libsync/src/sync.c', needed by 'obj/third_party/libsync/libsync/sync.o', missing and no known rule to make it

### Steps to Reproduce
New build

### Platform
Linux

### Full Command Used for Setup
./rattlesnakeos-stack --device walleye --name ros --prevent-shutdown --region us-east-2 --ssh-key us-east-2

### Email Notification output
Should look something like this:
```
RattlesnakeOS Build STARTED 
Device: walleye 
Stack Version: 0.0.15 
Build Date: 2018.07.06.15 
Elapsed Time: 0hrs 0min 6sec 
AOSP Build: OPM2.171026.006.G1 
AOSP Branch: android-8.1.0_r35 
Chromium Version: 68.0.3440.50 
F-Droid Version: 1.2.2 
F-Droid Priv Extension Version: 0.2.8
```

### Log output
N/A. Manually running build.sh. Last bit of lines:
Applying: kill Auth
Applying: kill remaining Play Services checks
error: patch failed: chrome/android/java_sources.gni:485
error: chrome/android/java_sources.gni: patch does not apply
error: patch failed: services/shape_detection/android/java/src/org/chromium/shape_detection/FaceDetectionProviderImpl.java:24
error: services/shape_detection/android/java/src/org/chromium/shape_detection/FaceDetectionProviderImpl.java: patch does not apply
Patch failed at 0001 kill remaining Play Services checks
The copy of the patch that failed is found in: .git/rebase-apply/patch
When you have resolved this problem, run "git am --continue".
If you prefer to skip this patch, run "git am --skip" instead.
To restore the original branch and stop patching, run "git am --abort".
Applying: kill TOS and metrics opt-out
Applying: kill Translate(doesn't work)
Installing Debian sid i386 root image: /home/ubuntu/chromium/src/build/linux/debian_sid_i386-sysroot
Downloading https://commondatastorage.googleapis.com/chrome-linux-sysroot/toolchain/1015a998c2adf188813cca60b558b0ea1a0b6ced/debian_sid_i386_sysroot.tar.xz
Installing Debian sid amd64 root image: /home/ubuntu/chromium/src/build/linux/debian_sid_amd64-sysroot
Downloading https://commondatastorage.googleapis.com/chrome-linux-sysroot/toolchain/1015a998c2adf188813cca60b558b0ea1a0b6ced/debian_sid_amd64_sysroot.tar.xz
Done. Made 22483 targets from 1498 files in 7466ms
ninja: Entering directory `out/Default/'
ninja: error: '../../third_party/libsync/src/sync.c', needed by 'obj/third_party/libsync/libsync/sync.o', missing and no known rule to make it

dan-v, 2018-07-06:
I'm going to pin to a known good release for time being so that I can stabilize build.

dan-v, 2018-07-06:
I've pinned Chromium to previous release for now in latest. Will leave this open until real fix is implemented.

dan-v, 2018-07-06:
Looks like it was trying to grab a beta release of Chromium and apply the existing patches. I've fixed this in the latest release. You can grab the latest (https://github.com/dan-v/rattlesnakeos-stack/releases) and update your stack (https://github.com/dan-v/rattlesnakeos-stack#how-to-update-rattlesnakeos-stack).

closed 2018-07-06 by dan-v