dan-v/rattlesnakeos-stack #16: Build deletes OTA after release if running two builds in one day

sjdrc, 2018-07-07:
### Prerequisites

* [ ] Are you running the latest version? Yes

### Description

After running the second build for a day (manually), the script uploads the ota files to aws, but in aws_release, it attempts to delete the old ota after uploading the new one, which results in the script deleting the file that it just uploaded.

Last lines of log
```
upload: release-taimen-2018.07.07.13/taimen-ota_update-2018.07.07.13.zip to s3://sjdrc-release/taimen-ota_update-2018.07.07.13.zip
delete: s3://sjdrc-release/taimen-ota_update-2018.07.07.13.zip
download: s3://sjdrc-release/taimen-target/taimen-target-files-2018.07.07.13.zip to ../taimen-target/taimen-target-files-2018.07.07.13.zip
~/rattlesnake-os/out ~/rattlesnake-os/out
~/rattlesnake-os/taimen-target ~/rattlesnake-os/out ~/rattlesnake-os/out
fatal error: An error occurred (404) when calling the HeadObject operation: Key "taimen-target/taimen-target-files-2018.07.07.00.zip" does not exist
delete: s3://sjdrc-release/taimen-target/taimen-target-files-2018.07.07.13.zip
upload: ../out/release-taimen-2018.07.07.13/taimen-target_files-2018.07.07.13.zip to s3://sjdrc-release/taimen-target/taimen-target-files-2018.07.07.13.zip
```

dan-v, 2018-07-07:
By manually, do you mean kicking it off from the Lambda function manually - or manually executing the script on an existing instance by yourself?

sjdrc, 2018-07-07:
Manually executing the build script, although judging from these lines:
```
old_date="$(cut -d ' ' -f 1 <<< "${old_metadata}")"
  (
  aws s3 cp "${BUILD_DIR}/out/release-${DEVICE}-${build_date}/${DEVICE}-ota_update-${build_date}.zip" "s3://${AWS_RELEASE_BUCKET}" --acl public-read &&
  echo "${build_date} ${build_timestamp} ${AOSP_BUILD}" | aws s3 cp - "s3://${AWS_RELEASE_BUCKET}/${RELEASE_CHANNEL}" --acl public-read &&
  echo "${BUILD_TIMESTAMP}" | aws s3 cp - "s3://${AWS_RELEASE_BUCKET}/${RELEASE_CHANNEL}-true-timestamp" --acl public-read
  ) && ( aws s3 rm "s3://${AWS_RELEASE_BUCKET}/${DEVICE}-ota_update-${old_date}.zip" || true )
```

This would occur regardless of the script execution method I think, due to old_date being the same as build_date.

dan-v, 2018-07-07:
If kicked off through Lambda a successful build should update the ${DEVICE}-stable file, and on next build it would pick up that date as old_date which shouldn't match the current date unless it was a build within the same hour which is unlikely unless you are kicking off multiple builds in the same hour. Definitely can be improved though and open to fixes.

sjdrc, 2018-07-07:
Ah right, I see. I've been refactoring a large portion of the build script, I'll make another issue to discuss the changes with you.

closed 2018-07-07 by sjdrc