dan-v/rattlesnakeos-stack #17: [ENHANCEMENT] Script refactor and redesign decisions

sjdrc, 2018-07-08:
Right now the build parts of the script is pretty closely coupled with the AWS release parts. I think that a better approach would be to generalise the build script so that it can be run on any machine or easily used in a systemd-nspawn or docker container by specifying environment variables. There seems to be a lack of simple, reliable build scripts for AOSP out there and this would be a useful resource for building a custom stack on another hosting site (for example digitalocean).

The script in `build_script.go` will then set the required build variables and then execute the build script.

I have already done a majority of the build script refactor and hope to have it up on a repo by the end of the weekend.

dan-v, 2018-07-08:
Heck yeah! Looking forward to seeing what you come up with.

dan-v, 2018-07-08:
@sjdrc - I think there is some good stuff there in terms of refactoring existing script and also some stuff I wouldn't pull into this project like signature spoofing patch. If I were to pull something like this into the project it would still need to maintain existing AWS functionality as I want this project to continue to be an end to end solution that sets up the entire stack with scheduled builds. As an intermediate step, I can see something like refactoring the existing script with environment variables so that it doesn't need to be generated in Go templating and could just be a raw shell script. You could for example, have an environment variable that turns on and off the AWS specific portions of the script if you were building locally or on some other platform. I'm also open to supporting other cloud platforms as well, but I would want it to be the same end to end experience that you get with AWS today.

sjdrc, 2018-07-09:
Yeah, I integrated signature spoofing for my own tests but the idea is that the applied patches will be easily customisable (In this case by inserting or adding a single line). I'd like to change it so that instead of just having the patches sequentially in a function, it load files from a scripts.d/ folder. This would mean it's easier for users to integrate changes they want without having to pull the script apart. I'm still running some more tests.

dan-v, 2018-07-21:
Going to close this one out. Happy to take improvements though as talked about above.

closed 2018-07-21 by dan-v