dan-v/rattlesnakeos-stack #20: Cannot login to Amazon Kindle

bunsim, 2018-07-21:
### Prerequisites

* [* ] Are you running the latest version?

### Description

Amazon Kindle doesn't login, claiming a problem with "connecting to Amazon". A webview with some sort of error momentarily pops up, probably meaning something wrong with the webview. I wonder whether this is due to the "privacy" patches done to Chromium preventing Kindle from displaying the Amazon stuff using the WebView correctly.

### Steps to Reproduce

1. Install Amazon Kindle (using something like Yalp Store)
2. Open Kindle
3. Attempt to log in

**Expected behavior:** WebView-based Amazon login screen appears

**Actual behavior:** WebView showing an error appears for a split seconds, and Kindle shows a toast saying that it cannot connect to Amazon.

bunsim, 2018-07-26:
This does indeed completely resolve the issue. Perhaps this should be documented? In any case I feel like the webview shouldn't be based on a adblocking, patched Chromium, though the default browser being that is okay. WebViews wouldn't really implement their API if they randomly block content, unlike the browser where the user is expected to have control over what content to browse, and many apps depend on exact behavior of WebViews.

dan-v, 2018-07-26:
I can understand that point of view, hence why I added an option to opt out. For me, I haven't had any issues with webview breakage in any of the apps I use and would prefer the additional privacy these patches provide. I'll try add something to the docs about opting out, or feel free to send a pull request to update docs.

closed 2018-07-26 by dan-v