dan-v/rattlesnakeos-stack #21: [Not a Bug] Building in an on-prem environment.

wertyy102, 2018-07-27:
Is it possible to build on your own hardware rather than AWS?

I have a 42u rack full of beefy CPU servers and plenty of RAM.  If not, would it be possible to work on a build that we can build in-house?

dan-v, 2018-07-27:
Unfortunately right now it is tied to AWS. I've had this request from a few different people though and usually point them at this project: https://github.com/aospbuild/aospbuild which covers the manual process of building plain AOSP w/ verified boot. Another option depending on how comfortable you are tearing apart shell scripts is to just take out all the AWS specific things from: https://github.com/dan-v/rattlesnakeos-stack/blob/master/templates/build_script.go

wertyy102, 2018-07-27:
Awesome.  I'll go tinker with the latter of the two and see what I can't make work.  Thanks!

closed 2018-07-27 by wertyy102