dan-v/rattlesnakeos-stack #22: Failed to create client: chmod C:\Users\User\AppData\Local\Temp\rattlesnakeos-stack850001234\terraform: The system cannot find the file specified.

captainhook, 2018-08-02:
### Prerequisites

* [X] Are you running the latest version?

### Description

On trying to deploy environment, I got this error.

### Steps to Reproduce

1. On CMD I ran "rattlesnakeos-stack.exe --region eu-west-1 --name rattlesnakeos-uniquename001 --device marlin --ssh-key privkey123.pem"

**Expected behavior:** Complete deployment.

**Actual behavior:**
[36mINFO[0m[0000] Checking AWS credentials
[36mINFO[0m[0001] Creating S3 bucket rattlesnakeos-uniquename001
[36mINFO[0m[0003] Downloading Terraform binary from URL: https://releases.hashicorp.com/terraform/0.11.7/terraform_0.11.7_windows_amd64.zip
[31mFATA[0m[0012] Failed to create client: chmod C:\Users\User\AppData\Local\Temp\rattlesnakeos-stack850001234\terraform: The system cannot find the file specified.

### Platform
Windows

### Full Command Used for Setup
rattlesnakeos-stack.exe --region eu-west-1 --name rattlesnakeos-uniquename001 --device marlin --ssh-key privkey123.pem

captainhook, 2018-08-02:
I've used Windows Subsystem for Linux and run the script inside Ubuntu, worked fine.

captainhook, 2018-08-02:
On a side note, @dan-v, were you able to build (using same signing keys, of course) and flash over your CopperheadOS installation without any issues?

dan-v, 2018-08-02:
Thanks for the report! I have never tried to actually run on Windows as I don't have a Windows box, so that is good to know. Glad you got it working with Windows subsystem for Linux!

I don't think it will be successful to flash over CopperheadOS. I'd recommend flashing factory image produced by the build.

closed 2018-08-02 by captainhook