dan-v/rattlesnakeos-stack #23: Install ConstraintLayout for Android 1.0.2" failed...

captainhook, 2018-08-02:
### Prerequisites

* [X] Are you running the latest version?

### Description

After patch_fdroid() runs, it seems to display an error running one of the other functions. It does however continue with rebuilding kernel.

**Actual behavior:**
Failed with an exception

### Platform
Ubuntu

### Full Command Used for Setup
./rattlesnakeos-stack --region us-west-2 --name rattlesnakeos-whatever --device marlin --ssh-key whatever

### Log output
==================================
```
Running apply_patches
==================================
~/rattlesnake-os/packages/apps/Updater/res/values ~/rattlesnake-os ~/chromium/src /
~/rattlesnake-os/packages/apps/F-Droid ~/rattlesnake-os/packages/apps/Updater/res/values ~/rattlesnake-os ~/chromium/src /
Downloading https://services.gradle.org/distributions/gradle-3.2.1-bin.zip
.................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
Unzipping /home/ubuntu/.gradle/wrapper/dists/gradle-3.2.1-bin/erlz51pt56t1o6vc7t39cikug/gradle-3.2.1-bin.zip to /home/ubuntu/.gradle/wrapper/dists/gradle-3.2.1-bin/erlz51pt56t1o6vc7t39cikug
Set executable permissions for: /home/ubuntu/.gradle/wrapper/dists/gradle-3.2.1-bin/erlz51pt56t1o6vc7t39cikug/gradle-3.2.1/bin/gradle
Starting a Gradle Daemon (subsequent builds will be faster)
Download https://jcenter.bintray.com/com/android/tools/build/gradle/2.2.2/gradle-2.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/build/gradle-core/2.2.2/gradle-core-2.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/build/builder/2.2.2/builder-2.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/lint/lint/25.2.2/lint-25.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/build/transform-api/2.0.0-deprecated-use-gradle-api/transform-api-2.0.0-deprecated-use-gradle-api.pom
Download https://jcenter.bintray.com/com/android/tools/build/gradle-api/2.2.2/gradle-api-2.2.2.pom
Download https://jcenter.bintray.com/com/android/databinding/compilerCommon/2.2.2/compilerCommon-2.2.2.pom
Download https://jcenter.bintray.com/org/ow2/asm/asm/5.0.4/asm-5.0.4.pom
Download https://jcenter.bintray.com/org/ow2/asm/asm-parent/5.0.4/asm-parent-5.0.4.pom
Download https://jcenter.bintray.com/org/ow2/ow2/1.3/ow2-1.3.pom
Download https://jcenter.bintray.com/org/ow2/asm/asm-commons/5.0.4/asm-commons-5.0.4.pom
Download https://jcenter.bintray.com/net/sf/proguard/proguard-gradle/5.2.1/proguard-gradle-5.2.1.pom
Download https://jcenter.bintray.com/net/sf/proguard/proguard-parent/5.2.1/proguard-parent-5.2.1.pom
Download https://jcenter.bintray.com/org/jacoco/org.jacoco.core/0.7.5.201505241946/org.jacoco.core-0.7.5.201505241946.pom
Download https://jcenter.bintray.com/org/jacoco/org.jacoco.build/0.7.5.201505241946/org.jacoco.build-0.7.5.201505241946.pom
Download https://jcenter.bintray.com/org/jacoco/org.jacoco.report/0.7.5.201505241946/org.jacoco.report-0.7.5.201505241946.pom
Download https://jcenter.bintray.com/net/sf/jopt-simple/jopt-simple/4.9/jopt-simple-4.9.pom
Download https://jcenter.bintray.com/org/sonatype/oss/oss-parent/7/oss-parent-7.pom
Download https://jcenter.bintray.com/com/google/protobuf/protobuf-java/2.5.0/protobuf-java-2.5.0.pom
Download https://jcenter.bintray.com/com/google/google/1/google-1.pom
Download https://jcenter.bintray.com/org/antlr/antlr/3.5.2/antlr-3.5.2.pom
Download https://jcenter.bintray.com/org/antlr/antlr-master/3.5.2/antlr-master-3.5.2.pom
Download https://jcenter.bintray.com/org/sonatype/oss/oss-parent/9/oss-parent-9.pom
Download https://jcenter.bintray.com/com/android/tools/build/builder-model/2.2.2/builder-model-2.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/build/builder-test-api/2.2.2/builder-test-api-2.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/sdklib/25.2.2/sdklib-25.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/sdk-common/25.2.2/sdk-common-25.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/common/25.2.2/common-25.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/build/manifest-merger/25.2.2/manifest-merger-25.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/ddms/ddmlib/25.2.2/ddmlib-25.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/jack/jack-api/0.11.0/jack-api-0.11.0.pom
Download https://jcenter.bintray.com/com/android/tools/jill/jill-api/0.10.0/jill-api-0.10.0.pom
Download https://jcenter.bintray.com/com/android/tools/analytics-library/protos/25.2.2/protos-25.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/analytics-library/shared/25.2.2/shared-25.2.2.pom
Download https://jcenter.bintray.com/com/android/tools/analytics-library/tracker/25.2.2/tracker-25.2.2.pom
Download https://jcenter.bintray.com/com/squareup/javawriter/2.5.0/javawriter-2.5.0.pom
Download https://jcenter.bintray.com/org/bouncycastle/bcpkix-jdk15on/1.48/bcpkix-jdk15on-1.48.pom
Download https://jcenter.bintray.com/org/bouncycastle/bcprov-jdk15on/1.48/bcprov-jdk15on-1.48.pom
Download https://jcenter.bintray.com/org/ow2/asm/asm-tree/5.0.4/asm-tree-5.0.4.pom
[... cut ...]
            project :app > com.github.pserwylo:BottomNavigation:1.5.0
      > Could not find com.android.support:design:25.3.1.
        Required by:
            project :app > com.github.pserwylo:BottomNavigation:1.5.0

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 45.461 secs
==================================
Running rebuild_marlin_kernel
==================================
Cloning into '/home/ubuntu/rattlesnake-os/kernel/google/marlin'...
```

dan-v, 2018-08-02:
The first build of F-Droid fails, but second during AOSP build should work. I have a to do note in build script to investigate this, but everything should still work in the end.

captainhook, 2018-08-02:
Thank you Dan!

I want to run a build again to see if it bumps into any more errors. If I was to force a new build, how would I do that? I see a FORCE_BUILD variable in the build script, so could I set that through Lambda?

dan-v, 2018-08-03:
There is a --force option you can pass to rattlesnakeos-stack tool and update your existing stack. Then kick off a manual build (documented in FAQ). Make sure to remove this --force option and apply that to your stack or you will keep doing forced builds going forward.

captainhook, 2018-08-03:
Awesome, thank you :)

closed 2018-08-03 by captainhook