dan-v/rattlesnakeos-stack #27: Resume chromium checkouts

Rudd-O, 2018-08-11:
I would like your program (the script template) that downloads and builds Chromium to please resume or at least reuse the Chromium checkout, because it takes like half an hour to download, and I've already hit the bandwidth cap for my IP address today from so much building.

Rudd-O, 2018-08-12:
```
	txt = strings.Replace(
		txt,
		"fetch --nohooks android || true",
		`fetch --nohooks android || {
	echo fetch failed, attempting gclient sync instead.
	gclient sync || {
		echo gclient sysn failed, attempting fresh fetch instead.
		cd ..
		rm -rf chromium
		mkdir -p chromium
		cd chromium
		fetch --nohooks android
	}
}`,
```

Rudd-O, 2018-08-13:
Updated snippet:

```
	txt = strings.Replace(
		txt,
		"fetch --nohooks android || true",
		`fetch --nohooks android || {
	echo fetch failed, attempting gclient sync instead.
	gclient sync && (
		for gitdir in $( find -name .git ) ; do
			pushd $gitdir/..
			git clean -fxd
			popd
		done
	) || {
		echo gclient sync failed, attempting fresh fetch instead.
		cd ..
		rm -rf chromium
		mkdir -p chromium
		cd chromium
		fetch --nohooks android
	} ; exit
}`,
```

dan-v, 2018-08-13:
If you are doing a strings.Replace on the script itself, I wouldn't recommend this approach. This will break anytime I make modifications. A better approach would be modifying the script so that it could handle both local and AWS builds by just changing a few parameters.

Anyways, happy to take a fully tested pull request for resuming chromium checkout if it already exists.

baldo, 2018-09-05:
There is `GIT_CACHE_PATH` that can be used to set a directory below which depot_tools will cache the repositories so those won't need a full re-download. In theory this should be equivalent to `git clone --mirror` into the cache directory and then clone / fetch from there. As I'm currently experimenting with this myself I ran into missing tag refs in the cache repos leading to errors on checkout. My current quick and dirty workaround is fetching the required tag on the cache repos before checkout:

```diff
   # checkout specific revision
+  if [[ -d "$GIT_CACHE_PATH" ]]; then
+    for dir in "$GIT_CACHE_PATH"/*; do
+      git \
+        -C "$dir" \
+        fetch origin refs/tags/"$CHROMIUM_REVISION":refs/tags/"$CHROMIUM_REVISION" || true
+    done
+  fi
+
   git checkout "$CHROMIUM_REVISION" -f
```

dan-v, 2018-09-05:
Closing this in favor of issue for supporting local builds: https://github.com/dan-v/rattlesnakeos-stack/issues/42

closed 2018-09-05 by dan-v