dan-v/rattlesnakeos-stack #29: Verify source code

Rudd-O, 2018-08-13:
`git verify-tag --raw $(git describe refs/tags/"$ANDROID_BUILD_TAG")` after `repo init`.

`repo forall -c 'git verify-tag --raw $(git describe)' || { ret=$? ; echo Verification failed! ; exit $ret ; }` after `repo sync`.

This allows us to verify that the code has in fact been signed.  Without this, there is no certainty that what you are building has not been tampered with.

Rudd-O, 2018-08-13:
```
repo init --manifest-url "$MANIFEST_URL" --manifest-branch "$AOSP_BRANCH" --depth 1 || true
...
...
...
# sync with retries
for i in {1..10}; do
  repo sync -c --no-tags --no-clone-bundle --jobs 32 && break
done
```

This code does not do any error checking.  If there is a problem here, downstream build should not continue.

Please remove the `||true` things.

dan-v, 2018-08-13:
1) There is nothing to verify at the moment as I haven't signed my git commits. I may do this in the future at some point. Opened issue #30 for this so it's not intermixed with other issues.
2) '|| true' is there for errors that I don't care about, which is this case for this one as if repo is already initialized I don't want to error out and exit the script. The repo sync will fail anyways if repo init was not successful. There surely is a more elegant way to handle this and open to fully tested PR for improvements.

closed 2018-08-13 by dan-v

dan-v, 2018-08-13:
You are importing Daniel's pgp key and verifying tags signed by him. What I am saying is that my commits and tags are not signed, so this will not work.

The AOSP pgp key (documented here https://source.android.com/setup/build/downloading) which is probably the other key you have in your repo can be used to verify tags for repos in default manifest which covers most but not everything.

Rudd-O, 2018-08-13:
Ignore Daniel's key.  That obviously has nothing to do with this.  What I mean is that your code should use the GPG verification with the AOSP key.

Also, this error:

```
remote: Daily bandwidth rate limit exceeded for IP 85.195.241.225
fatal: protocol error: bad pack header
```

Is glossed over by the `|| true` that you've added.  This trick only causes a harder to debug build failure later.  It's better to just fail outright, or at least if the tenth sync retry has failed, then fail.  At the moment, your program does not do either.

I'll be working around these issues in my build scripts, and later I may send a PR with the fixes.  There are quite a few things I've had to work around so far, but it's looking good.