dan-v/rattlesnakeos-stack #33: userdebug and eng builds

Rudd-O, 2018-08-20:
https://github.com/dan-v/rattlesnakeos-stack/blob/c717345338ded07ccdcf93edb974b394f053db2e/templates/build_script.go#L42

`user` should probably be a variable that also acccepts `userdebug` or `eng` (useful to get a build that will support ADB out of the box to diagnose boot failures).

chrfischer, 2018-08-20:
Dear Rudd-O,

wouldn't it be more polite to offer well tested Pull requests instead of opening several issues containing single lines of code?

Rudd-O, 2018-08-20:
Indeed it would be.  We'd have to discuss in private why that's not possible.

dan-v, 2018-08-21:
Hey @Rudd-O - appreciate all of the potential improvements listed in the issues you've opened, but I would definitely like to see this as a single fully tested PR. Feel free to message me on reddit - if you need to chat offline about why that's not possible.

dan-v, 2018-08-21:
In 9.0 branch I've made this a separate variable.

closed 2018-08-21 by dan-v