dan-v/rattlesnakeos-stack #37: Fails to build cannot obtain manifest

Rudd-O, 2018-08-21:
This is on a squeaky-clean checkout.  Still won't work.  I'm using master branch, not 9.0 branch.

```
+ echo 'Running fetch_aosp_source'
Running fetch_aosp_source
+ echo ==================================
==================================
+ pushd /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/rattlesnake-os
~/rattlesnake-os ~
+ repo init --manifest-url https://android.googlesource.com/platform/manifest --manifest-branch AOSP_BRANCH --depth 1
gpg: keybox '/home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/.repoconfig/gnupg/pubring.kbx' created
gpg: /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/.repoconfig/gnupg/trustdb.gpg: trustdb created
gpg: key 16530D5E920F5C65: public key "Repo Maintainer <repo@android.kernel.org>" imported
gpg: key 67B7E448692B382C: public key "Conley Owens <cco3@android.com>" imported
gpg: Total number processed: 2
gpg:               imported: 2
Get https://gerrit.googlesource.com/git-repo/clone.bundle
Get https://gerrit.googlesource.com/git-repo
Get https://android.googlesource.com/platform/manifest
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Server does not provide clone.bundle; ignoring.
fatal: Couldn't find remote ref refs/heads/AOSP_BRANCH
fatal: Couldn't find remote ref refs/heads/AOSP_BRANCH
fatal: cannot obtain manifest https://android.googlesource.com/platform/manifest
+ true
+ grep -q RattlesnakeOS .repo/manifest.xml
grep: .repo/manifest.xml: No such file or directory
+ awk -i inplace -v FDROID_CLIENT_VERSION=1.3.1 -v FDROID_PRIV_EXT_VERSION=0.2.8 '1;/<repo-hooks in-project=/{
      print "  ";
      print "  <remote name=\"github\" fetch=\"https://github.com/RattlesnakeOS/\" revision=\"master\" />";
      print "  <remote name=\"fdroid\" fetch=\"https://gitlab.com/fdroid/\" />";
      print "  <remote name=\"prepare-vendor\" fetch=\"https://github.com/anestisb/\" revision=\"master\" />";
      print "  ";
      print "  <project path=\"script\" name=\"script\" remote=\"github\" />";
      print "  <project path=\"external/chromium\" name=\"platform_external_chromium\" remote=\"github\" />";
      print "  <project path=\"packages/apps/Updater\" name=\"platform_packages_apps_Updater\" remote=\"github\" />";
      print "  <project path=\"packages/apps/F-Droid\" name=\"fdroidclient\" remote=\"fdroid\" revision=\"refs/tags/" FDROID_CLIENT_VERSION "\" />";
      print "  <project path=\"packages/apps/F-DroidPrivilegedExtension\" name=\"privileged-extension\" remote=\"fdroid\" revision=\"refs/tags/" FDROID_PRIV_EXT_VERSION "\" />";
      print "  <project path=\"vendor/android-prepare-vendor\" name=\"android-prepare-vendor\" remote=\"prepare-vendor\" />"}' .repo/manifest.xml
awk: inplace:25: warning: inplace_begin: Cannot stat `.repo/manifest.xml' (No such file or directory)
awk: inplace:25: fatal: cannot open file `.repo/manifest.xml' for reading (No such file or directory)
```

dan-v, 2018-08-21:
Looks like you may have some modifications in place as the master branch should be using a variable $AOSP_BRANCH (not AOSP_BRANCH) that is dynamically determined from AOSP website.

https://github.com/dan-v/rattlesnakeos-stack/blob/9.0/templates/build_template.go#L314

dan-v, 2018-08-21:
Verified on master this does not happen. Closing this out.

closed 2018-08-21 by dan-v

Rudd-O, 2018-08-22:
You were right.