dan-v/rattlesnakeos-stack #38: Pixel build fail on upgrade to Android P

jamesocIRL, 2018-08-23:
### Prerequisites

* [Y ] Are you running the latest version?

### Description
Build fails on sailfish.

### Steps to Reproduce

1. Download latest stack and execture.
2. Email informing me it has failed

**Expected behavior:** Build

**Actual behavior:** Failed

### Platform
Linux
### Full Command Used for Setup
AWS_ACCESS_KEY_ID=??? AWS_SECRET_ACCESS_KEY=??? ./rattlesnakeos-stack --region us-west-2 --name rattlej1 --device sailfish
### Carrier
Cricket

### Email Notification output

From
AWS Notifications <no-reply@sns.amazonaws.com>
Th 23. Aug 02:14
Sender
01010165648f71d9-ca2649b1-c783-4703-8f9a-386a80f0088e-000000@us-west-2.amazonses.com
To
Me <jamesoc@tuta.io>
AWS Notification Message
RattlesnakeOS Build FAILED
Device: sailfish
Stack Name: rattlej1
Stack Version: 9.0.2
Release Channel: sailfish-stable
Build Date: 2018.08.22.21
Elapsed Time: 4hrs 58min 56sec
AOSP Build: PPR1.180610.009
AOSP Branch: android-9.0.0_r1
Chromium Version: 69.0.3497.41
F-Droid Version: 1.3.1
F-Droid Priv Extension Version: 0.2.8
et/product/sailfish/obj/STATIC_LIBRARIES/libv8gen_intermediates/experimental-extra-libraries.cc
[ 1% 1437/86805] Copy: out/target/product/sailfish/obj/STATIC_LIBRARIES/libv8gen_intermediates/extra-libraries.cc
[ 1% 1438/86805] bc: libclcore_neon.bc_32 <= frameworks/rs/driver/runtime/arch/clamp.c
[ 1% 1439/86805] Check module type: out/target/product/sailfish/obj/SHARED_LIBRARIES/libpac_intermediates/link_type
[ 1% 1440/86805] Check module type: out/target/product/sailfish/obj/SHARED_LIBRARIES/libnfc_nci_jni_intermediates/link_type
[ 1% 1441/86805] Generating experimental-libraries.cc
[ 1% 1442/86805] Copy: out/target/product/sailfish/obj/STATIC_LIBRARIES/libv8gen_intermediates/experimental-libraries.cc
[ 1% 1443/86805] Check module type: out/target/product/sailfish/obj/APPS/NfcNci_intermediates/jni_link_type
[ 1% 1444/86805] Check module type: out/target/product/sailfish/obj/SHARED_LIBRARIES/libjni_pacprocessor_intermediates/link_type
[ 1% 1445/86805] Check module type: out/target/product/sailfish/obj/APPS/PacProcessor_intermediates/jni_link_type
[ 1% 1446/86805] host C++: libv8base <= external/v8/src/base/once.cc
[ 1% 1447/86805] Generating libraries.cc
[ 1% 1448/86805] Copy: out/target/product/sailfish/obj/STATIC_LIBRARIES/libv8gen_intermediates/libraries.cc
[ 1% 1449/86805] Export includes file: out/target/product/sailfish/obj/STATIC_LIBRARIES/libv8gen_intermediates/libraries.cc -- out/target/product/sailfish/obj/STATIC_LIBRARIES/libv8gen_intermediates/export_includes
[ 1% 1450/86805] host C++: libv8base <= external/v8/src/base/cpu.cc
[ 1% 1451/86805] host C++: libv8base <= external/v8/src/base/functional.cc
[ 1% 1452/86805] host C++: v8mkpeephole <= external/v8/src/interpreter/bytecode-operands.cc
[ 1% 1453/86805] host C++: libv8base <= external/v8/src/base/file-utils.cc
[ 1% 1454/86805] host C++: libv8base <= external/v8/src/base/division-by-constant.cc
[ 1% 1455/86805] host C++: libv8base <= external/v8/src/base/debug/stack_trace.cc
[ 1% 1456/86805] host C++: libv8base <= external/v8/src/base/logging.cc
[ 1% 1457/86805] host C++: libv8base <= external/v8/src/base/platform/mutex.cc
[ 1% 1458/86805] Check module type: out/host/linux-x86/obj/EXECUTABLES/v8mkpeephole_intermediates/link_type
[ 1% 1459/86805] host C++: libv8base <= external/v8/src/base/ieee754.cc
[ 1% 1460/86805] build out/target/common/obj/PACKAGING/hiddenapi-light-greylist.txt
[... 263 lines cut ...]
  @libcore.util.NonNull (nl=false) @ {GENERIC_ARRAY_LOCATION=GenericArrayLocationCriterion at outermost type, RETURN_TYPE=ReturnTypeCriterion for method: append(C)Ljava/lang/Appendable;, IN_METHOD=in method 'append(C)Ljava/lang/Appendable;', IN_CLASS=In class 'java.io.PrintWriter' (exactly)}
Found class java.io.PrintWriter, but unable to insert @libcore.util.NonNull:
  @libcore.util.NonNull (nl=false) @ {GENERIC_ARRAY_LOCATION=GenericArrayLocationCriterion at outermost type, RETURN_TYPE=ReturnTypeCriterion for method: append(Ljava/lang/CharSequence;)Ljava/io/Writer;, IN_METHOD=in method 'append(Ljava/lang/CharSequence;)Ljava/io/Writer;', IN_CLASS=In class 'java.io.PrintWriter' (exactly)}
Found class java.io.PrintWriter, but unable to insert @libcore.util.Nullable:
  @libcore.util.Nullable (nl=false) @ {GENERIC_ARRAY_LOCATION=GenericArrayLocationCriterion at outermost type, PARAM=ParamCriterion for method: append(Ljava/lang/CharSequence;) at position: 0, IN_METHOD=in method 'append(Ljava/lang/CharSequence;)Ljava/io/Writer;', IN_CLASS=In class 'java.io.PrintWriter' (exactly)}
Found class java.io.PrintWriter, but unable to insert @libcore.util.NonNull:
  @libcore.util.NonNull (nl=false) @ {GENERIC_ARRAY_LOCATION=GenericArrayLocationCriterion at outermost type, RETURN_TYPE=ReturnTypeCriterion for method: append(Ljava/lang/CharSequence;)Ljava/lang/Appendable;, IN_METHOD=in method 'append(Ljava/lang/CharSequence;)Ljava/lang/Appendable;', IN_CLASS=In class 'java.io.PrintWriter' (exactly)}
Found class java.io.PrintWriter, but unable to insert @libcore.util.Nullable:
  @libcore.util.Nullable (nl=false) @ {GENERIC_ARRAY_LOCATION=GenericArrayLocationCriterion at outermost type, PARAM=ParamCriterion for method: append(Ljava/lang/CharSequence;) at position: 0, IN_METHOD=in method 'append(Ljava/lang/CharSequence;)Ljava/lang/Appendable;', IN_CLASS=In class 'java.io.PrintWriter' (exactly)}
Found class java.io.PrintWriter, but unable to insert @libcore.util.NonNull:
  @libcore.util.NonNull (nl=false) @ {GENERIC_ARRAY_LOCATION=GenericArrayLocationCriterion at outermost type, RETURN_TYPE=ReturnTypeCriterion for method: append(Ljava/lang/CharSequence;II)Ljava/io/Writer;, IN_METHOD=in method 'append(Ljava/lang/CharSequence;II)Ljava/io/Writer;', IN_CLASS=In class 'java.io.PrintWriter' (exactly)}
Found class java.io.PrintWriter, but unable to insert @libcore.util.Nullable:
  @libcore.util.Nullable (nl=false) @ {GENERIC_ARRAY_LOCATION=GenericArrayLocationCriterion at outermost type, PARAM=ParamCriterion for method: append(Ljava/lang/CharSequence;II) at position: 0, IN_METHOD=in method 'append(Ljava/lang/CharSequence;II)Ljava/io/Writer;', IN_CLASS=In class 'java.io.PrintWriter' (exactly)}
Found class java.io.PrintWriter, but unable to insert @libcore.util.NonNull:
  @libcore.util.NonNull (nl=false) @ {GENERIC_ARRAY_LOCATION=GenericArrayLocationCriterion at outermost type, RETURN_TYPE=ReturnTypeCriterion for method: append(Ljava/lang/CharSequence;II)Ljava/lang/Appendable;, IN_METHOD=in method 'append(Ljava/lang/CharSequence;II)Ljava/lang/Appendable;', IN_CLASS=In class 'java.io.PrintWriter' (exactly)}
Found class java.io.PrintWriter, but unable to insert @libcore.util.Nullable:
  @libcore.util.Nullable (nl=false) @ {GENERIC_ARRAY_LOCATION=GenericArrayLocationCriterion at outermost type, PARAM=ParamCriterion for method: append(Ljava/lang/CharSequence;II) at position: 0, IN_METHOD=in method 'append(Ljava/lang/CharSequence;II)Ljava/lang/Appendable;', IN_CLASS=In class 'java.io.PrintWriter' (exactly)}
Found class java.util.HashMap, but unable to insert @libcore.util.NonNull:
  @libcore.util.NonNull (nl=false) @ {GENERIC_ARRAY_LOCATION=GenericArrayLocationCriterion at ( [TYPE_ARGUMENT(0)] ), RETURN_TYPE=ReturnTypeCriterion for method: entrySet()Ljava/util/Set;, IN_METHOD=in method 'entrySet()Ljava/util/Set;', IN_CLASS=In class 'java.util.HashMap' (exactly)}
Found class java.util.HashMap, but unable to insert @libcore.util.NullFromTypeParam:
  @libcore.util.NullFromTypeParam (nl=false) @ {GENERIC_ARRAY_LOCATION=GenericArrayLocationCriterion at ( [TYPE_ARGUMENT(0), TYPE_ARGUMENT(0)] ), RETURN_TYPE=ReturnTypeCriterion for method: entrySet()Ljava/util/Set;, IN_METHOD=in method 'entrySet()Ljava/util/Set;', IN_CLASS=In class 'java.util.HashMap' (exactly)}
Found class java.util.HashMap, but unable to insert @libcore.util.NullFromTypeParam:
  @libcore.util.NullFromTypeParam (nl=false) @ {GENERIC_ARRAY_LOCATION=GenericArrayLocationCriterion at ( [TYPE_ARGUMENT(0), TYPE_ARGUMENT(1)] ), RETURN_TYPE=ReturnTypeCriterion for method: entrySet()Ljava/util/Set;, IN_METHOD=in method 'entrySet()Ljava/util/Set;', IN_CLASS=In class 'java.util.HashMap' (exactly)}
ninja: build stopped: subcommand failed.
02:14:00 ninja failed with: exit status 1

dan-v, 2018-08-23:
At first glance it looks like a network blip trying to download gradle. I'll take a further look later but trying again by kicking off another build manually may fix this.

dan-v, 2018-08-27:
Gonna close this out. Let me know if retrying with another build doesn't fix the issue.

closed 2018-08-27 by dan-v

jamesocIRL, 2018-08-28:
Appology for the delay in getting back to you. Been a busy week. It fails upon rebuild.

RattlesnakeOS Build FAILED
Device: sailfish
Stack Name: rattlej1
Stack Version: 9.0.2
Release Channel: sailfish-stable
Build Date: 2018.08.24.22
Elapsed Time: 0hrs 24min 28sec
AOSP Build: PPR1.180610.009
AOSP Branch: android-9.0.0_r1
Chromium Version: 69.0.3497.53
F-Droid Version: 1.3.1
F-Droid Priv Extension Version: 0.2.8
% 66.3M 0s
77500K .......... .......... .......... .......... .......... 90% 58.4M 0s
77550K .......... .......... .......... .......... .......... 90% 67.6M 0s
77600K .......... .......... .......... .......... .......... 90% 51.6M 0s
77650K .......... .......... .......... .......... .......... 90% 69.9M 0s
77700K .......... .......... .......... .......... .......... 90% 59.8M 0s
77750K .......... .......... .......... .......... .......... 90% 53.6M 0s
77800K .......... .......... .......... .......... .......... 90% 68.0M 0s
77850K .......... .......... .......... .......... .......... 90% 51.4M 0s
77900K .......... .......... .......... .......... .......... 90% 58.1M 0s
77950K .......... .......... .......... .......... .......... 90% 106M 0s
78000K .......... .......... .......... .......... .......... 90% 49.2M 0s
78050K .......... .......... .......... .......... .......... 90% 77.6M 0s
78100K .......... .......... .......... .......... .......... 90% 70.5M 0s
78150K .......... .......... .......... .......... .......... 90% 73.5M 0s
78200K .......... .......... .......... .......... .......... 90% 46.1M 0s
78250K .......... .......... .......... .......... .......... 90% 54.2M 0s
78300K .......... .......... .......... .......... .......... 91% 81.0M 0s
78350K .......... .......... .......... .......... .......... 91% 60.1M 0s
78400K .......... .......... .......... .......... .......... 91% 27.9M 0s
78450K .......... .......... .......... .......... .......... 91% 75.6M 0s
78500K .......... .......... .......... .......... .......... 91% 52.8M 0s
78550K .......... .......... .......... .......... .......... 91% 87.0M 0s
78600K .......... .......... .......... .......... .......... 91% 63.1M 0s
78650K .......... .......... .......... .......... .......... 91% 35.8M 0s
78700K .......... .......... .......... .......... .......... 91% 49.5M 0s
78750K .......... .......... .......... .......... .......... 91% 77.0M 0s
78800K .......... .......... .......... .......... .......... 91% 36.3M 0s
78850K .......... .......... .......... .......... .......... 91% 72.2M 0s
78900K .......... .......... .......... .......... .......... 91% 76.8M 0s
78950K .......... .......... .......... .......... .......... 91% 64.7M 0s
79000K .......... .......... .......... .......... .......... 91% 67.9M 0s
79050K .......... .......... .......... .......... .......... 91% 61.4M 0s
79100K .......... .......... .......... .......... .......... 91% 75.4M 0s
79150K .......... .......... .......... .......... .......... 91% 57.1M 0s
79200K .......... .......... .......... .......... .......... 92% 51.7M 0s
79250K .......... .......... .......... .......... .......... 92% 49.7M 0s
79300K .......... .......... .......... .......... .......... 92% 64.0M 0s
79350K .......... .......... .......... .......... .......... 92% 65.6M 0s
79400K .......... .......... .......... .......... .......... 92% 64.3M 0s
79450K .......... .......... .......... .......... .......... 92% 83.0M 0s
79500K .......... .......... .......... .......... .......... 92% 84.7M 0s
79550K .......... .......... .......... .......... .......... 92% 57.9M 0s
79600K .......... .......... .......... .......... .......... 92% 55.9M 0s
79650K .......... .......... .......... .......... .......... 92% 51.0M 0s
79700K .......... .......... .......... .......... .......... 92% 79.6M 0s
79750K .......... .......... .......... .......... .......... 92% 56.5M 0s
79800K .......... .......... .......... .......... .......... 92% 45.4M 0s
79850K .......... .......... .......... .......... .......... 92% 59.5M 0s
79900K .......... .......... .......... .......... .......... 92% 36.8M 0s
79950K .......... .......... .......... .......... .......... 92% 7.79M 0s
80000K .......... .......... .......... .......... .......... 92% 52.9M 0s
80050K .......... .......... .......... .......... .......... 93% 74.8M 0s
80100K .......... .......... .......... .......... .......... 93% 86.9M 0s
80150K .......... .......... .......... .......... .......... 93% 45.9M 0s
80200K .......... .......... .......... .......... .......... 93% 40.5M 0s
[... 207 lines cut ...]
udev 15G 0 15G 0% /dev
tmpfs 3.0G 39M 3.0G 2% /run
/dev/xvda1 194G 55G 140G 29% /
tmpfs 15G 0 15G 0% /dev/shm
tmpfs 5.0M 0 5.0M 0% /run/lock
tmpfs 15G 0 15G 0% /sys/fs/cgroup
/dev/loop0 87M 87M 0 100% /snap/core/4830
/dev/loop1 13M 13M 0 100% /snap/amazon-ssm-agent/295
53G /home/ubuntu/rattlesnake-os
53G total
23:06:13 up 25 min, 0 users, load average: 1.51, 2.02, 2.15
Completed 256.0 KiB/3.2 MiB with 1 file(s) remaining
Completed 512.0 KiB/3.2 MiB with 1 file(s) remaining
Completed 768.0 KiB/3.2 MiB with 1 file(s) remaining
Completed 1.0 MiB/3.2 MiB with 1 file(s) remaining
Completed 1.2 MiB/3.2 MiB with 1 file(s) remaining
Completed 1.5 MiB/3.2 MiB with 1 file(s) remaining
Completed 1.8 MiB/3.2 MiB with 1 file(s) remaining
Completed 2.0 MiB/3.2 MiB with 1 file(s) remaining
Completed 2.2 MiB/3.2 MiB with 1 file(s) remaining
Completed 2.5 MiB/3.2 MiB with 1 file(s) remaining
Completed 2.8 MiB/3.2 MiB with 1 file(s) remaining
Completed 3.0 MiB/3.2 MiB with 1 file(s) remaining
Completed 3.2 MiB/3.2 MiB with 1 file(s) remaining
upload: ../../../../../../../../var/log/cloud-init-output.log to s3://rattlej1-logs/sailfish/1535151973

dan-v, 2018-08-28:
Wow, another networking connectivity issue - this time just during a 'git clone' of a repo. I'm not sure why you are having such bad luck with network connectivity.  I don't currently have any retry logic in place to account for network issues (as it hasn't been an issue up until this point), so for now unfortunately the best option I can give you is to just retry the build again. I've also opened https://github.com/dan-v/rattlesnakeos-stack/issues/39 to take a look at adding retries to network related operations. If you do decide to build again, you'll need to upgrade to latest 9.0.3 release in order to avoid a Chromium build error.