dan-v/rattlesnakeos-stack #42: Add support for local builds

dan-v, 2018-09-05:
In the current state the tool is AWS specific. It would be nice to have a method of supporting local builds without any dependency on AWS.

lrvick, 2018-11-12:
I got building in docker working which will make it possible to build locally or remote with one command, and target a wider range of build and CI/CD systems and make deterministic builds easier.

https://github.com/hashbang/os

The idea is keep the build tool image take in a given configs/$device.env and return deterministic flashable images.

Might want a similar approach here.

Currently can build/flash crosshatch but touchscreen driver issues atm...

dan-v, 2018-11-12:
I'm happy to pull any tooling changes that simplify the build process, but I still want it to function as a full end to end solution at the end of the day. But I think for people looking to do local builds this is a great start. 

Also I'm still interested in deterministic builds, but as we chatted about previously it doesn't mesh well with the current setup which is targeted at full automation and is always grabbing the latest versions which means the build is inherently not deterministic at the moment.

lrvick, 2018-11-13:
I am starting from the other side of the problem domain but I think we still share a similar endgame.

For automated builds my plan is to wrap these containers in a helm chart where the new build profile discovery service will run multiple times per day looking for new potential profiles, then will attempt to build them and publish signed results as a "latest" release channel. From there when other CI systems confirm the same hashes we can get a "verified" release channel. Looking to basically do a setup similar to bitcoin core in that respect.

Containers/k8s will make the entire stack portable to run locally (minikube) or on digitalocean, aws, azure, or any other provider with k8s support.

The secondary goal is for the container to generate deterministic builds provided it gets the same config.env. A separate service can be responsible for automatically discovering new config.env configs which then CI systems can build and determine they all get the same hash and issue signatures.

I think there is a path to have our cake and eat it too... but we will see how far I get!

I'll be stealing from current rattlesnakeos heavily ;)

dan-v, 2018-11-14:
I think the end game is a little different between the two projects, but that's not a bad thing at all. I want this project to be more about a cost effective simplified tool with end to end automation for building a personalized AOSP for your device with privacy and security in mind. It sounds like your project is trying to create a sane production quality AOSP build system that more advanced users could setup and also share images with less technical users in a secure manner - rather than just "hey bro, install this binary on your phone - trust me".

Anyways, I really do think this is a great idea and would love to see it up and running in the future. I'd potentially be interested in helping, but I would like to see some solid progress towards these goals. I was excited initially about this idea when we talked a few months ago, but my interest fell off pretty quickly when there wasn't anyone making progress. I do know how real life can get in the way of side projects like these. Feel free to steal what you need - that's what it is there for and I'll be keeping an eye on your progress.

lrvick, 2018-11-15:
Deterministic builds for me also include reproducible infrastructure. My intent is that people can use hashbang signed builds and update server, -or- deploy their own using a couple commands. I want people deploying their own and then we can all cross-check each others hashes.

I am leaning very strongly towards a Helm chart as an alternative to all the terraform/cloudformation stuff to gain support for all hosting providers as well as local building with the same toolchain. That said your solution is probably as cheap as it gets and could also be adapted to a container based build triggered by lambda as well. Probably getting ahead of myself :)

Anywho. I realized quickly I would need to buy a new phone to really be able to build/test all this.

Now with no updates on my copperheados phone I bought a pixel 3 to be the replacement which means I had to pull the trigger on going all in on this and put some other projects on hold.

Gotta get that setup and automatic updating to become my daily driver... so a forcing function.

Life indeed :)

dan-v, 2018-11-16:
I think a containerized approach with kubernetes is good and wrapping that up in a helm chart will help. 

I can definitely relate to the forcing function of a project like this. Having to get things working so your phone is usable :) I ended up getting a beatup second phone off Ebay to do testing with, so that my primary phone stopped being broken all the time.

yassineim, 2019-01-29:
You might be interested in this: https://github.com/localstack/localstack

Rudd-O, 2019-03-13:
I would just https://github.com/Rudd-O/rattlesnakeos-build .  There are local and Jenkins build modes now.

We can collaborate to reduce the amount of patching I have to do to your program, and I'm confident the amount of patching can be brought down to zero, if we end up merging my custom build template renderer into your codebase (which will probably then compile to a separate binary, something like `rattlesnakeos-local` in addition to `rattlesnakeos-stack`).

To ease porting, I can independently maintain the Jenkins portion on my own repo, no need to merge that.

dan-v, 2019-03-13:
Hi @Rudd-O. We've chatted previously on this topic a while ago, and while I'm glad to see you've managed to get local builds and Jenkins builds working, it's not using a mechanism that I'd consider supportable, especially the [render portion](https://github.com/Rudd-O/rattlesnakeos-build/blob/master/render.go) of your code that is rip and replacing things from the template. Below are the items I'm looking for in order to do a proper PR for local builds that is supportable moving forward. I think you have a bunch of these things already and certainly further along than any other local builds I've seen, but just would need a bit more working to properly integrate it.

- Refactor rattlesnakeos-stack to have a local build mode (something like 'rattlesnakeos-stack local <flags etc>' which bypasses all AWS specific logic. Local build mode could either 1) just spit out a generated shell script that could be executed locally or 2) have rattlesnakeos-stack execute the generated shell script locally.

- Separate out the AWS specific logic in the template script. The current script does many operations with S3 for example that would need to be replaced with local filesystem copies/reads/writes. The quick and dirty solution would just be a flag variable in shell script to denote local builds (e.g. LOCAL=true/false) and then just sprinkle if statements everywhere. It would probably be cleaner to use the go templating to generate a shell script with only the local or AWS specific logic in it.

- There's also a bunch of pieces of the build script that are currently not designed to be rerun which would need to be cleaned up in order run locally without having to do things like pull down the full AOSP or Chromium source code on every build for example. These changes to make it idempotent for local builds would need to obviously not break anything for AWS builds.

- If the OTA updater was still to be functional in local build mode, the OTA specific metadata would have to be dumped to local filesystem, the URL for the update endpoint would need to be configurable so you could for example run a local webserver for updates or maybe you still want to put your updates on S3 or another public endpoint. This update endpoint would likely need to be an additional flag in rattlesnakeos-stack that gets plumbed into the shell script template.

lrvick, 2019-03-13:
>    like 'rattlesnakeos-stack local ' which bypasses all AWS specific logic.

FYI #!os yeilds essentially the same images as rattlesnakeos now based on
whatever config file you feed in.

https://github.com/hashbang/os

Wrapping this with a similar automated-spot-instance-build bootstrapping
like dan-v has done here would not be super difficult, then you can have
your local cake and eat it automated too :)

Rudd-O, 2019-03-14:
> Hi @Rudd-O. We've chatted previously on this topic a while ago, and while I'm glad to see you've managed to get local builds and Jenkins builds working, it's not using a mechanism that I'd consider [...]

You and I agree.  The plan would be to merge most of those changes and refactor the unmergeable ones, so that they become formally supported.

> Below are the items I'm looking for in order to do a proper PR for local builds that is supportable moving forward. I think you have a bunch of these things already and certainly further along than any other local builds I've seen, but just would need a bit more working to properly integrate it.

I'd be happy to collaborate with you.

I think the next step is for me to begin submitting small PRs that tackle each patch, starting with those that do *not* affect AWS builds at all.  I agree with all your points.

The OTA updater is not supposed to work in local mode, only in Jenkins mode.  That said, the OTA updater release URL is in fact specifiable in local mode, to support the case of people who SCP files from their build machine into their personal Web server.  I plan to enhance the documentation to include information on how to do that.  I believe local mode in my program already dumps the metadata in the directory `s3/rattlesnakeos-release` which makes it easy to just deploy files + metadata via `scp` or `rsync`.

Rudd-O, 2019-03-14:
@lrvick Not to critizise, but to point out: your program's source code looks drastically different from RattlesnakeOS. — it's an entirely different beast.  Which file implements support for RattlesnakeOS configs?  In my program, it's `render.go`.

dan-v, 2019-03-14:
I think what @lrvick has created is a much cleaner, well thought out AOSP build implementation in comparison to just the giant shell script template that is RattlesnakeOS. Having a docker based build also allows it to be much more portable which is very useful for local builds. RattlesnakeOS build process was designed purely around getting a build done in AWS, so it's a lot less flexible. 

@Rudd-O - I think the approach of small PRs that tackle each patch, starting with those that do not affect AWS builds at all is great. I don't want to directly merge into 9.0 branch, and instead I'll setup a 'local' branch for these small PRs with the goal of eventually merging that into the main branch. Let's see where that takes us.

ypid, 2019-03-20:
Hi @Rudd-O 

I have reviewed and tested https://github.com/hashbang/os over the last months and really like it. Thanks to @lrvick and friends for the awesome work! Also thanks very much to @dan-v for the well maintained Rattlesnake OS!

closed 2022-03-05 by dan-v