dan-v/rattlesnakeos-stack #45: Build Failing

cryptocub3, 2018-09-07:
### Prerequisites

* [ ] Are you running the latest version?

Yes

### Description
Build stops before finish

[Description of the bug or feature]

Here is a part of /var/log/syslog witch I think is relevant

```
Sep  7 18:09:44 ip-172-31-43-140 cloud-init[2261]: Preparing to unpack .../bsdiff_4.3-15_amd64.deb ...
Sep  7 18:09:44 ip-172-31-43-140 cloud-init[2261]: Unpacking bsdiff (4.3-15) ...
Sep  7 18:09:44 ip-172-31-43-140 cloud-init[2261]: Processing triggers for man-db (2.7.5-1) ...
Sep  7 18:09:44 ip-172-31-43-140 cloud-init[2261]: Setting up bsdiff (4.3-15) ...
Sep  7 18:09:46 ip-172-31-43-140 cloud-init[2261]: install-build-deps-android.sh complete.
Sep  7 18:09:46 ip-172-31-43-140 cloud-init[2261]: Not applying any patches to Chromium as requested
Sep  7 18:09:52 ip-172-31-43-140 cloud-init[2261]: Done. Made 22505 targets from 1535 files in 6432ms
Sep  7 18:09:52 ip-172-31-43-140 cloud-init[2261]: ninja: Entering directory `out/Default/'
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]: ninja: error: '../../third_party/proguard/lib/proguard.jar', needed by 'gen/chrome/android/webapk/libs/runtime_library/webapk_runtime_library.jar', missing and no known rule to make it
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]: Filesystem      Size  Used Avail Use% Mounted on
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]: udev             15G     0   15G   0% /dev
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]: tmpfs           3.0G   17M  3.0G   1% /run
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]: /dev/xvda1      194G   42G  152G  22% /
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]: tmpfs            15G     0   15G   0% /dev/shm
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]: tmpfs           5.0M     0  5.0M   0% /run/lock
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]: tmpfs            15G     0   15G   0% /sys/fs/cgroup
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]: /dev/loop0       87M   87M     0 100% /snap/core/4830
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]: /dev/loop1       13M   13M     0 100% /snap/amazon-ssm-agent/295
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]: 4.0K#011/home/ubuntu/rattlesnake-os
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]: 4.0K#011total
Sep  7 18:09:54 ip-172-31-43-140 cloud-init[2261]:  18:09:54 up 28 min,  0 users,  load average: 1.33, 2.01, 2.03
Sep  7 18:10:09 ip-172-31-43-140 cloud-init[2261]: Completed 88.0 KiB/953.7 KiB with 1 file(s) remaining#015upload failed: ../../../../var/log/cloud-init-output.log to s3://rattlesnakeos-private-logs/walleye/1536343794 ('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer'))
```

### Platform
Linux

### Full Command Used for Setup

./rattlesnakeos-stack --remove --region us-west-2 --name rattlesnakeos-private --device walleye --ssh-key ...

### Email Notification output
No event triggered

### Log output
No log uploaded

dan-v, 2018-09-08:
Ah yes, sorry for confusion. I've been working only on 9.0 branch and have set that to primary branch on Github.

cryptocub3, 2018-09-08:
Got it working thanks. Just a simple if you don't mind answering what is rattlesnake os updater system app ? As I understand all updates are working trough sideloading what is that ?