dan-v/rattlesnakeos-stack #46: Skip chromium build?

vvug, 2018-09-13:
It seems that building Chromium is an important fraction of the total build time. I'm going to use Bromite as a browser anyway (it integrates an adblock and other stuff). Is it possible to skip building Chromium? It would be nice to have a flag for that.

dan-v, 2018-09-13:
The idea behind building Chromium is not necessarily to use it as your primary browser (use whatever your favorite browser is) but in order to provide a secure and up to date webview. Webview is a primary system component of Android that is used by many applications to display web components. The standard AOSP webview [is quite old](https://android.googlesource.com/platform/external/chromium-webview/+log/android-9.0.0_r1/prebuilt/arm64) in comparison. Before the 9.0 release, I was building Chromium with patches from Bromite (which also applied to the webview), but unfortunately it 1) caused too many build issues for me to continue to support right now 2) caused issues with some applications that were using webview.

vvug, 2018-09-13:
And if I remove the AOSP webview and skip building Chromium I guess Bromite can't register as a webview provider after I install it from the apk...

dan-v, 2018-09-13:
Yeah, you definitely need a webview implementation for Android to work - almost every app uses a webview in some form. Bromite does provide a system webview as part of it's release artifacts that could be downloaded to the build tree and configured as the webview as part of build process. I'd still prefer not to trust a primary component of the Android OS by pulling in this binary blob though, which is why I was just taking the patches from Bromite and applying it to the corresponding Chromium version and doing the build myself. I may toy with the idea of locking to a version of Chromium for a period of time or allowing users to specify the version of Chromium they want, as it does take a very long time to build.

vvug, 2018-09-13:
Then I'm closing this. Thanks for your feedback.

closed 2018-09-13 by vvug