dan-v/rattlesnakeos-stack #48: In email notification provide reason why build is required

dan-v, 2018-09-16:
In email notification, highlight reason why the build is required (e.g. Chromium version changed, etc).

pgera, 2018-11-18:
I'm seeing some unnecessary builds. This would definitely help in debugging. I just got a notification for a build although nothing changed in upstream. Does it also build if the stack's version is updated ? Secondly, it hasn't even been 14 days since my last build, which is what the config file has.

dan-v, 2018-11-18:
Yes, stack version change will cause build to continue. The only thing that should cause it quicker than 14 days is if that value was modified at some point as that ends up recreating the cloudwatch event cron under the covers. This also has some weird behavior as it will also kick off a build when the schedule is updated. One way to work around this would be to define a cron schedule with exact day of month and time you want to run rather than just weeks like I have currently. I may switch over to that approach anyways as it has less weird side effects.

pgera, 2018-11-18:
Does a manual build affect the schedule ? Ideally it should. The clock should reset to the last successful build. Or at least, that seems natural for most use cases. I don't know if there is any benefit in sticking to the schedule despite a recent manual build.

dan-v, 2018-11-18:
Manual build does not affect schedule, its essentially just like going to the UI and running the Lambda function. Agreed something like that would be more natural, but would need to think a bit about how to best implement.

dan-v, 2018-11-19:
A very simplistic 'Build Reason' field has been added to email notifications in v9.0.16.

closed 2018-11-19 by dan-v