dan-v/rattlesnakeos-stack #49: Fails to build because sdk license isn't accepted

pgera, 2018-09-23:
### Prerequisites

* [x] Are you running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases)?

### Description

Fails to build because sdk license isn't accepted

### Platform
Linux

### Full Command Used for Setup
./rattlesnakeos-stack --region us-east-1 --instance-type c5.9xlarge --name rattlesnakeos-pg --device sailfish --ssh-key keyfile

### Email Notification output
Should look something like this:
```
RattlesnakeOS Build FAILED
  Device: sailfish
  Stack Name: rattlesnakeos-pg
  Stack Version: 9.0.6 
  Stack Region: us-east-1
  Release Channel: sailfish-stable
  Instance Type: c5.9xlarge
  Instance Region: us-east-2
  Build Date: 2018.09.23.04
  Elapsed Time: 2hrs 13min 48sec
  AOSP Build: PPR2.180905.006
  AOSP Branch: android-9.0.0_r7
  Chromium Version: 69.0.3497.100
  F-Droid Version: 1.4
  F-Droid Priv Extension Version: 0.2.8
/sf/proguard/proguard-base/5.3.3/proguard-base-5.3.3.pom
Download https://jcenter.bintray.com/com/google/code/findbugs/jsr305/1.3.9/jsr305-1.3.9.pom
Download https://jcenter.bintray.com/com/google/j2objc/j2objc-annotations/1.1/j2objc-annotations-1.1.pom
Download https://jcenter.bintray.com/org/codehaus/mojo/animal-sniffer-annotations/1.14/animal-sniffer-annotations-1.14.pom
Download https://jcenter.bintray.com/org/codehaus/mojo/animal-sniffer-parent/1.14/animal-sniffer-parent-1.14.pom
Download https://jcenter.bintray.com/org/codehaus/mojo/mojo-parent/34/mojo-parent-34.pom
Download https://jcenter.bintray.com/org/codehaus/codehaus-parent/4/codehaus-parent-4.pom
Download https://maven.google.com/com/android/tools/dvlib/26.1.1/dvlib-26.1.1.pom
Download https://maven.google.com/com/android/tools/repository/26.1.1/repository-26.1.1.pom
Download https://maven.google.com/com/android/tools/layoutlib/layoutlib-api/26.1.1/layoutlib-api-26.1.1.pom
Download https://jcenter.bintray.com/org/apache/httpcomponents/httpmime/4.1/httpmime-4.1.pom
Download https://jcenter.bintray.com/org/apache/commons/commons-compress/1.12/commons-compress-1.12.pom
Download https://jcenter.bintray.com/org/apache/httpcomponents/httpclient/4.2.6/httpclient-4.2.6.pom
Download https://jcenter.bintray.com/org/apache/httpcomponents/httpcomponents-client/4.1/httpcomponents-client-4.1.pom
Download https://jcenter.bintray.com/org/apache/commons/commons-parent/39/commons-parent-39.pom
Download https://jcenter.bintray.com/org/apache/httpcomponents/httpcomponents-client/4.2.6/httpcomponents-client-4.2.6.pom
Download https://jcenter.bintray.com/org/apache/httpcomponents/project/4.1.1/project-4.1.1.pom
Download https://jcenter.bintray.com/org/apache/httpcomponents/project/7/project-7.pom
Download https://jcenter.bintray.com/org/apache/apache/16/apache-16.pom
Download https://jcenter.bintray.com/org/apache/apache/13/apache-13.pom
Download https://jcenter.bintray.com/net/sf/kxml/kxml2/2.3.0/kxml2-2.3.0.pom
Download https://jcenter.bintray.com/org/jetbrains/annotations/13.0/annotations-13.0.pom
Download https://jcenter.bintray.com/org/glassfish/jaxb/jaxb-runtime/2.2.11/jaxb-runtime-2.2.11.pom
Download https://jcenter.bintray.com/com/sun/activation/javax.activation/1.2.0/javax.activation-1.2.0.pom
Download https://jcenter.bintray.com/com/google/jimfs/jimfs/1.1/jimfs-1.1.pom
Download https://jcenter.bintray.com/com/google/jimfs/jimfs-parent/1.1/jimfs-parent-1.1.pom
Download https://jcenter.bintray.com/com/sun/activation/all/1.2.0/all-1.2.0.pom
Download https://jcenter.bintray.com/com/sun/xml/bind/mvn/jaxb-runtime-parent/2.2.11/jaxb-runtime-parent-2.2.11.pom
Download https://jcenter.bintray.com/net/java/jvnet-parent/1/jvnet-parent-1.pom
Download https://jcenter.bintray.com/com/sun/xml/bind/mvn/jaxb-parent/2.2.11/jaxb-parent-2.2.11.pom
Download https://jcenter.bintray.com/com/sun/xml/bind/jaxb-bom-ext/2.2.11/jaxb-bom-ext-2.2.11.pom
Download https://jcenter.bintray.com/org/glassfish/jaxb/jaxb-bom/2.2.11/jaxb-bom-2.2.11.pom
Download https://jcenter.bintray.com/net/java/jvnet-parent/4/jvnet-parent-4.pom
Download https://jcenter.bintray.com/commons-codec/commons-codec/1.6/commons-codec-1.6.pom
Download https://jcenter.bintray.com/org/apache/httpcomponents/httpcore/4.2.5/httpcore-4.2.5.pom
Download https://jcenter.bintray.com/commons-logging/commons-logging/1.1.1/commons-logging-1.1.1.pom
Download https://jcenter.bintray.com/org/apache/httpcomponents/httpcomponents-core/4.2.5/httpcomponents-core-4.2.5.pom
Download https://jcenter.bintray.com/org/apache/commons/commons-parent/5/commons-parent-5.pom
[... 188 lines cut ...]
Completed 7.5 MiB/12.1 MiB with 1 file(s) remaining  
Completed 7.8 MiB/12.1 MiB with 1 file(s) remaining  
Completed 8.0 MiB/12.1 MiB with 1 file(s) remaining  
Completed 8.2 MiB/12.1 MiB with 1 file(s) remaining  
Completed 8.5 MiB/12.1 MiB with 1 file(s) remaining  
Completed 8.8 MiB/12.1 MiB with 1 file(s) remaining  
Completed 9.0 MiB/12.1 MiB with 1 file(s) remaining  
Completed 9.2 MiB/12.1 MiB with 1 file(s) remaining  
Completed 9.5 MiB/12.1 MiB with 1 file(s) remaining  
Completed 9.8 MiB/12.1 MiB with 1 file(s) remaining  
Completed 10.0 MiB/12.1 MiB with 1 file(s) remaining 
Completed 10.2 MiB/12.1 MiB with 1 file(s) remaining 
Completed 10.5 MiB/12.1 MiB with 1 file(s) remaining 
Completed 10.8 MiB/12.1 MiB with 1 file(s) remaining 
Completed 11.0 MiB/12.1 MiB with 1 file(s) remaining 
Completed 11.2 MiB/12.1 MiB with 1 file(s) remaining 
Completed 11.5 MiB/12.1 MiB with 1 file(s) remaining 
Completed 11.8 MiB/12.1 MiB with 1 file(s) remaining 
Completed 12.0 MiB/12.1 MiB with 1 file(s) remaining 
Completed 12.1 MiB/12.1 MiB with 1 file(s) remaining 
upload: ../../../var/log/cloud-init-output.log to s3://rattlesnakeos-pg-logs/sailfish/1537685453
```

### Log output
Log file is quite big. Can attach if needed. (Also not sure if all sensitive information is pruned from the log)

dan-v, 2018-09-23:
I haven't seen this issue before. If you could grab from the logs the section labeled 'Running setup_env' it would help debug the issue as this is where the SDK is setup and agreed to. I did kick of a new build on my end just to see if it's a persistent problem, and I wasn't able to reproduce. Kicking off another build manually will likely result in a successful build, but if you can get the logs for that section it would help to see the error and hopefully I can make it a bit more robust.

pgera, 2018-09-23:
Attaching the log. From the logs, it looks like the sdk license is accepted. Not sure what's the reason for the error. I'll try launching another build. 

[1537685453.txt](https://github.com/dan-v/rattlesnakeos-stack/files/2409024/1537685453.txt)

pgera, 2018-09-23:
The build finished on the next attempt

pgera, 2018-09-24:
I am not able to flash the image though

```
extracting boot.img (29 MB) to disk... took 0.049s
archive does not contain 'boot.sig'
archive does not contain 'boot_other.img'
archive does not contain 'dtbo.img'
archive does not contain 'dt.img'
archive does not contain 'odm.img'
archive does not contain 'product.img'
archive does not contain 'product-services.img'
archive does not contain 'recovery.img'
archive does not contain 'super.img'
extracting system.img (985 MB) to disk... took 1.323s
archive does not contain 'system.sig'
extracting system_other.img (90 MB) to disk... took 0.126s
archive does not contain 'system.sig'
archive does not contain 'vbmeta.img'
extracting vendor.img (259 MB) to disk... took 0.344s
archive does not contain 'vendor.sig'
archive does not contain 'vendor_other.img'
/usr/local/bin/mke2fs failed with status 1
fastboot: error: Cannot generate image for userdata
```

dan-v, 2018-09-24:
Just want to make sure you are following the prerequisite advice from flashing guide as you didn't mention version of tools being used: 

You should have at least 4GB of memory on your OS to avoid problems.

You can obtain the adb and fastboot tools from the Android SDK. Either install Android Studio or use the standalone SDK. Do not use distribution packages for adb and fastboot. Distribution packages are out-of-date and not compatible with the latest version of Android. An obsolete fastboot will result in corrupted installations and potentially bricked devices. Do not make the common mistake of assuming that everything will be fine and ignoring these instructions. Double check that the first fastboot in your PATH is indeed from an up-to-date SDK installation:

pgera, 2018-09-24:
adb and fastboot were fine. They were from google's latest platform tools. However, I have never had to use mke2fs from google's platform tools while flashing stock images.

dan-v, 2018-09-24:
If you are using google's latest platform tools then it should be using the provided mke2fs in ./platform-tools/mke2fs. Is it possible you also have installed the tools through your package manager and maybe that it is using tools from there?

    ➜  ~ find ./sdk -name mke2fs
    ./sdk/platform-tools/mke2fs

pgera, 2018-09-24:
I had copied over fastboot and adb from google's platform tools to /usr/local/bin. I never needed the other tools included from platform-tools for flashing images. However, I always used to run the final "fastboot -w update" without the -w flag since I never updated the userdata. So that explains why it never complained earlier. So it's fine. Everything is working as expected.

dan-v, 2018-09-24:
Ahh, gotcha! Let me know how it works out.

pgera, 2018-09-24:
It's been working great. Couple of minor things that I noticed: The location icon is always present at the top right, instead of it coming on when an application actually requests location. I guess this is an AOSP limitation ? I don't remember if CoS also had this issue though. Do you remember ? Second, there isn't an PDF viewer in this build. I'm using the mupdf reader from f-droid. It's working fine. Is this what everyone uses ? I don't think CoS's pdf reader is still maintained, right ?

dan-v, 2018-09-24:
I believe the location icon is always present because the GPS is set to device only. I'm not positive if this also happened in CoS, never really paid attention to it. As for your second question, I'm trying to avoid adding any additional apps to RattlesnakeOS, as I don't want the additional overhead of maintaining it, and I'd rather have people choose what apps they want anyways. I use Dropbox for backing up photos and I've just been using their PDF viewer, but mupdf looks like a better FOSS option.

dan-v, 2018-09-25:
I wasn't able to find the root cause in the logs unfortunately - it does appear it accepted the SDK agreement. If this issue pops up again I'll try to investigate further.

closed 2018-09-25 by dan-v