dan-v/rattlesnakeos-stack #52: Add new feature to allow additional pre-built APKs to be included in builds

dan-v, 2018-10-04:
With a new optional flag, allow a git repo url to be provided which contains pre-built APKs and the required Android makefile to include it into the build process.

stevesoltys, 2018-10-05:
This sounds great.

The ability to provide a set of patches would be useful as well. We'd need to fetch additional build scripts and execute them during `apply_patches`. The order would matter, so maybe it would make sense to order them alphabetically and use the standard patch format - `0001-my-patch.sh`.

dan-v, 2018-10-06:
agreed @stevesoltys! I've made an additional tracking item for doing patches in this same fashion #54.

dan-v, 2018-10-08:
I've released an initial version of this feature in latest release. Added new flag --repo-prebuilts that allows users to provide a git repo that contains additional prebuilt APK files to include in build. This is mostly a prototype to get some feedback as I'm not sure what is the best approach here.

thatchrisblack, 2018-10-31:
Would this possible allow those of us stuck on Sprint to be able install Rattlesnake successfully? Though I admit I'm not certain what's different about Ting on their network than a postpaid Sprint account. :) Is it SprintDM that's the issue?

dan-v, 2018-10-31:
@thatchrisblack - not a high priority item for me, but I've opened up an issue for supporting Sprint (https://github.com/dan-v/rattlesnakeos-stack/issues/69).

thatchrisblack, 2018-10-31:
Thanks! No hurry. New to Android dev work and was mostly curious over what Sprint requires that other CDMA carriers don't. :)

dan-v, 2018-11-04:
Closing this out, although this behavior may change in a future release if I decide to use alternative local manifest like in issue #60.

closed 2018-11-04 by dan-v