dan-v/rattlesnakeos-stack #54: Add new feature to allow additional patches to be included in builds

dan-v, 2018-10-06:
"The ability to provide a set of patches would be useful as well. We'd need to fetch additional build scripts and execute them during apply_patches. The order would matter, so maybe it would make sense to order them alphabetically and use the standard patch format - 0001-my-patch.sh."

dan-v, 2018-10-08:
I've released an initial version of this feature in latest release. Added new flag --repo-patches that allows users to provide a git repo that contains additional patches to apply to AOSP build root. This is mostly a prototype to get some feedback as I'm not sure what is the best approach here.

stevesoltys, 2018-10-08:
LGTM - since the patches are applied in the order specified by the manifest. I think that's most important since, for example, the sensors permission patch will likely depend on the internet permission patch (they [use the same function](https://github.com/AndroidHardeningArchive/platform_frameworks_base/commit/a2d7dbf5520a85dd42ab6d5da90c9d3391f476d0#diff-cb589fb4d249734569e11adf2d1c645aR2214) in the original CopperheadOS feature).

dan-v, 2018-10-08:
@stevesoltys - two questions/asks for you:
- If you could take a quick look at the initial implementation of the patching (https://github.com/dan-v/rattlesnakeos-stack/blob/9.0/templates/build_template.go#L468) and let me know if that would work or not it would be super helpful. Basically just defining the list of patches in a manifest file which ultimately decides the ordering they are applied.
- You'll also see I modified your patch filenames (https://github.com/RattlesnakeOS/community_patches/blob/9.0/00001-global-internet-permission-toggle.patch) to instead be from the root of the AOSP build directory rather than from within the specific git repo, as this allows patching of pretty much anything, but at the same time I'm wondering if it might be a pain for creating the patches. If painful, any idea on making this better?

stevesoltys, 2018-10-08:
Sure @dan-v.

1. I think the manifest looks good. However, I was originally thinking that it could be nice to allow bash scripts for patching. You could source any bash scripts that are provided in the manifest so that they could use tools like sed, and any environmental variables from the build script. For example, [something like this](https://github.com/dan-v/rattlesnakeos-stack/blob/9.0/templates/build_template.go#L571). Patches written like that may be less likely to break when code is changed upstream. Could allow both options?

2. I think that's fine for me, and if additional bash scripts are allowed as mentioned above the community could write patches in the same format as they are currently.

dan-v, 2018-10-08:
I like that idea of supporting scripts. I'll have to think about the best way to support both patches and scripts here.

dan-v, 2018-10-12:
@stevesoltys - latest release now has support for shell scripts as patches as well. Basically just looking at the file extension for each file defined in the manifest to determine if it should either run patch or execute shell script. An example that changes boot animation: https://github.com/RattlesnakeOS/example_patch_shellscript.

stevesoltys, 2018-10-12:
@dan-v looks great, thanks!

Slightly related, I tried porting the sensors patch to Android P. While the permission appears, it doesn't seem to actually be restricting the app's ability to use sensors. I'll need to look into it deeper, it may be less trivial than expected. I'll open a PR on the community patches repo to discuss progress on that.

stevesoltys, 2018-10-12:
Nevermind, I know why. I'll open that PR once I get it working.

dan-v, 2018-10-12:
Awesome! Would love to have that as well!

dan-v, 2018-11-04:
Closing this out, although this behavior may change in a future release if I decide to use alternative local manifest like in issue #60.

closed 2018-11-04 by dan-v