dan-v/rattlesnakeos-stack #56: Bootloop with image built from latest code

Rudd-O, 2018-10-09:
### Prerequisites

* [x] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [x] I am able to reproduce this issue without any advanced customization options like --repo-patches or --repo-apks flags

### Description

Image flashed from marlin-factory-latest.tar.xz/flash-all.sh bootloops and eventually gets to recovery.

How do I upload the full build log?

dan-v, 2018-10-09:
@Rudd-O - you didn't include the majority of the details requested (version, command run, email output, etc), which are included in the template to help debug issues. I also have to ask if this is your modified version or standard latest version v9.0.6+.

Rudd-O, 2018-10-09:
This is modified, the build process is _identical_.  I can show you my script, give you a diff with the build_template.go file, everything.  Will disclose everything now, save for the massive 80 MB log.

Rudd-O, 2018-10-09:
My script:

```

#!/bin/bash

if [ $# -ne 1 ]; then
  echo "Need to specify device name as argument"
  exit 1
fi

# check if supported device
DEVICE=$1
if [ "$DEVICE" == 'sailfish' ] || [ "$DEVICE" == 'marlin' ] || [ "$DEVICE" == 'walleye' ] || [ "$DEVICE" == 'taimen' ]; then
  echo "Supported device $DEVICE - continuing build"
else 
  echo "Unsupported device $DEVICE"
  exit 1
fi

# set region
REGION=none
export AWS_DEFAULT_REGION=${REGION}

# stack name
STACK_NAME=rattlesnakeos

# version of stack running
STACK_VERSION=

# prevent default action of shutting down on exit
PREVENT_SHUTDOWN=

# force build even if no new versions exist of components
FORCE_BUILD=false

# skip chromium build if there is an existing build
SKIP_CHROMIUM_BUILD=false

# pin to specific version of android
ANDROID_VERSION="9.0"

# build type (user or userdebug)
BUILD_TYPE="user" # replaced

# build channel (stable or beta)
BUILD_CHANNEL="stable"

# user customizable things
REPO_PATCHES=
REPO_PREBUILTS=
HOSTS_FILE=

# get current instance details for reference
INSTANCE_TYPE=none
INSTANCE_REGION=none
INSTANCE_IP=none

# aws settings
AWS_KEYS_BUCKET="${STACK_NAME}-keys"
AWS_RELEASE_BUCKET="${STACK_NAME}-release"
AWS_LOGS_BUCKET="${STACK_NAME}-logs"
AWS_SNS_ARN=none

# build settings
BUILD_TARGET="release aosp_${DEVICE} ${BUILD_TYPE}"
RELEASE_URL="https://${AWS_RELEASE_BUCKET}.s3.amazonaws.com"
RELEASE_CHANNEL="${DEVICE}-${BUILD_CHANNEL}"
CHROME_CHANNEL="stable"
BUILD_DATE=$(date +%Y.%m.%d.%H)
[... 1219 lines cut ...]
+    get_latest_versions
+    check_for_new_versions
+    aws_notify "RattlesnakeOS Build STARTED"
+    setup_env
+    check_chromium
+    fetch_aosp_source
+    setup_vendor
+    aws_import_keys
+    apply_patches
+    # only marlin and sailfish need kernel rebuilt so that verity_key is included
+    if [ "${DEVICE}" == "marlin" ] || [ "${DEVICE}" == "sailfish" ]; then
+      rebuild_marlin_kernel
+    fi
+    build_aosp
+    aws_release
+    checkpoint_versions
+    aws_notify "RattlesnakeOS Build SUCCESS"
+  fi
+}
+fi
+
+RELEASE_URL=http://example.com/
 full_run
-`
```

Rudd-O, 2018-10-09:
I stress this was built with a 100% clean workspace directory

Rudd-O, 2018-10-09:
Version:  `> git checkout -f d45d3a49b0989497841390d593518b3611e6710e`

dan-v, 2018-10-09:
I'm sure it's very similar to the original, but there is likely some difference causing issues here as I've done a successful flash as of yesterday for marlin. I don't at the moment have time to dig in the differences between the two build scripts, but if I have some free cycles I'll take a peak.

Usually you get into a boot loop state without loading the boot animation screen when there is an issue with the kernel. Unfortunately that also means you can't get really any output through ADB to troubleshoot unless you have a debug cable. I'd double check you are properly building the kernel and also including the appropriate verity key with the kernel.

Hopefully you didn't lock the bootloader before doing any of this experimentation as this means you can easily get out of the situation. If you have the bootloader locked, you are basically in a situation where you now need to produce a properly booting image signed with the same key to flash on your device.

Rudd-O, 2018-10-09:
Yes, absolutely no boot animation screen.

Fortunately for me, my phone is both unlocked and OEM unlocked right now -- it's sitting next to my keyboard, idling in the recovery screen.

Perhaps you can furnish me with a build logfile of one of your builds, so that I can diff it with mine, and see if there's anything different?

Rudd-O, 2018-10-09:
In the meantime, here's the kernel stage build log.
[consoleText kernel build.log](https://github.com/dan-v/rattlesnakeos-stack/files/2462205/consoleText.kernel.build.log)

dan-v, 2018-10-09:
[1539069428.txt](https://github.com/dan-v/rattlesnakeos-stack/files/2462301/1539069428.txt)

Rudd-O, 2018-10-09:
Man, I see no meaningful differences in our kernel build.   About the only things different are the order of compilation of the `.o` files.  Will continu to investigate.

Rudd-O, 2018-10-10:
I need to get a console-ramoops dump of the machine while it is running.  How do I build a recovery image that will not actually bootloop back to recovery when something goes wrong?

Rudd-O, 2018-10-10:
I believe I have found the bug.  My script is more aggressive than yours in cleaning junk from source directories.  Since, in my edition, the step `fetch_aosp_source` resets the `device/google/marlin-kernel` repository to its original state, I have to suppress the copy of the kernel in `rebuild_marlin_kernel` and move it to `fetch_aosp_source` (where I also integrate the Chromium APK).

The reason we did not catch this, is that (unlike the Chromium prebuilt APKs git repo) the kernel directory actually has a file `Image.lz4-dtb`, so there is never a _lack_ of kernel that would trigger a build failure.

There are too many side effects not tracked by version control in this pipeline.  We should collaborate to reduce them.

My edition of the program also does out-of-tree kernel builds now.  This seriously improves the certainty that the kernel is being built pristinely and only with whatever patches your program may contain, never confused by `.gitignore`.

Rudd-O, 2018-10-10:
(Will report soon.)

Rudd-O, 2018-10-10:
Yeah, I've verified my code actually builds in the new kernel, but to no avail.  As a last ditch attempt, I will attempt another full clean build with the code updates.

Rudd-O, 2018-10-10:
I found this in the ramoops log:

```
[    1.882916] c3      1 ALSA device list:
[    1.882922] c3      1   No soundcards found.
[    1.882974] c3      1 Warning: unable to open an initial console.
[    1.882995] c3      1 dm: attempting early device configuration.
[    1.883485] c3      1 device-mapper: init: adding target '0 1 android-verity 259:17    '
[    1.883498] c3      1 device-mapper: android-verity: key:id:5d0936bec3848ab90c5dfaa8c80ab1f4f9dd5068 dev:259:17
[    1.887540] c3      1 device-mapper: android-verity: bio magic_number:2952900609 protocol_version:0 table_length:266
[    1.887551] c3      1 device-mapper: android-verity: verity_table: 1 /dev/block/platform/soc/624000.ufshc/by-name/system /dev/block/platform/soc/624000.ufshc/by-name/system 4096 4096 516099 516099 sha256 a1beaf398c2d163e3ce048fa3a69dc25948d9227371ba5b44df712611ac061ad aee087a5be3b982978c923f566a94613496b417f2af592639bc80d141e34dfe7
[    1.887570] c3      1 device-mapper: android-verity: keyring: key not found
[    1.887575] c3      1 device-mapper: android-verity: Signature verification failed
[    1.887585] c3      1 device-mapper: android-verity: triggering restart
[    1.887603] c3      1 htc_reboot_block reboot_block: reason=77665508 msg=dm-verity device corrupted save to /misc+00003800
[    1.887624] c3      1 htc_reboot_block reboot_block: unable to open file: /dev/block/platform/soc/624000.ufshc/by-name/misc
[    1.887629] c3      1 [BATT] reboot_consistent_command_call: save batt cycle data
[    1.887633] c3      1 [BATT] reboot_consistent_command_call: save consistent data
[    1.887638] c3      1 [BATT] change_level_by_consistent_and_store_into_emmc: Battery data not ready, don't save consistent data.
[    1.887643] c3      1  at 1887627895 (1970-02-01 07:58:06.530014065 UTC)
[    1.887653] c2     58 [TP]: [ERR]:fwu_start_reflash: Firmware image synaptics_bl71.img not available
```

Is this why the thing restarts?

dan-v, 2018-10-10:
Indeed that looks like the issue.

Rudd-O, 2018-10-10:
Goddammit.  I'm positive this is another problem with my code.  Stand by 16 hours for more info...

dan-v, 2018-10-10:
Hopefully you get it figured out. 16 hours is a painful turn around time :( On the c5.9xlarge instances I can do a full rebuild in 1 hour now which is pretty nice.

Rudd-O, 2018-10-10:
I might just dispatch test builds directly on Amazon AWS, as my build server is already fully-equipped to dispatch such builds -- just a bit of tweaking on the Jenkins pipeline and that should do it.

Rudd-O, 2018-10-10:
(The problem is I'm not willing to flash them on my phone.  I'm gonna have to get another one just to test on it.)

Rudd-O, 2018-10-10:
Hm, key theft is a problem, but not the only one -- machines not under your control can inject malicious code more easily than those under your control.  (It's probably obvious that I'm not a big fan of "other people's computers").

dan-v, 2018-10-10:
Yeah that's a bit beyond my threat model for sure, but I get it.

stevesoltys, 2018-10-10:
FWIW I ran into this issue (that being a kernel boot loop) after uploading my previous CopperheadOS keys to the S3 bucket and doing a build. The solution was to have the build script generate the keys - I'm not sure if this is related but I had the same symptoms. It didn't make sense to me as to why that would happen but it seems similar given the issue is related to verity.

Rudd-O, 2018-10-10:
Eugh.  This is scary.

dan-v, 2018-10-10:
Yeah, there's a [thread on reddit ](https://reddit.com/r/RattlesnakeOS/comments/96eu5e/has_anyone_tested_upgrading_from_copperheados_to/) where I dove into the reasons why you can't install RattlesnakeOS over the top of CopperheadOS and some potential ways to workaround it.

Rudd-O, 2018-10-10:
OMG OMG.  I think I know what the problem is!

stevesoltys, 2018-10-10:
Right @dan-v, but the odd thing was the factory image had the same issue, not just flashing the OTA over CopperheadOS. That's why I thought it might be relevant here.

Rudd-O, 2018-10-10:
Flashing the new image.  This time the build log said this:

```
  CERTS   kernel/x509_certificate_list
  - Including cert /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/rattlesnake-os/kernel/google/marlin/verifiedboot_marlin_relkeys.der.x509
  - Including cert /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/rattlesnake-os/kernel/google/marlin/verifiedboot_sailfish_relkeys.der.x509
  - Including cert /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/rattlesnake-os/kernel/google/marlin/verity_dev_keys.der.x509
  - Including cert /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/rattlesnake-os/kernel/google/marlin/verity_rdev_keys.der.x509
  - Including cert /home/user/jenkins/workspace/RattlesnakeOS_images_master-ROOFFD3KKAISMC4G76UPIWCPKBIHZ73OD3D77HHSZLNDKO3FBUSA/rattlesnake-os/keys/marlin/verity_user.der.x509
```

Let's hope that does the trick.

Rudd-O, 2018-10-10:
IT WORKED!  Thank you so much guys!

closed 2018-10-10 by Rudd-O