dan-v/rattlesnakeos-stack #57: df during the android-prepare-vendor hangs it

pgera, 2018-10-11:
I think I'm hitting this: https://sourceforge.net/p/fuse-ext2/discussion/787602/thread/762bcae2/?limit=25

ssh to the instance. If you call df in the prepare-vendor stage, df will hang, and prepare-vendor also hangs.
The log has been stuck at this for a while:
```
[*] Processing standalone symlinks
[*] Processing additional runtime generated product files from APK symlinks
[*] Generating signatures file
[*] Generating build_id file
```
Btw, isn't the prepare-vendor output saved to s3 ? Since I'm redoing a build, this should have ideally been skipped.

dan-v, 2018-10-11:
I've definitely bumped into this issue before (although usually not on standard clean builds). Unfortunately all of the tools that exist to extract these images on linux are not in the best state. fuse-ext2 is not very robust, but of the options provided by android-prepare-vendor (debugfs, fuse-ext2, ext4fuse), it was the only one I could get working in the existing build environment. I'm currently working on upgrading the build OS from Ubuntu 16.04 to 18.04 and it appears the newer version of debugfs allows for more reliable extraction, so I'll probably swap out fuse-ext2 with debugfs on the next release. I'll also add an enhancement issue to investigate having timeouts on operations like this, so builds don't hang until the spot instance expires (12 hours right now). Right now prepare-vendor is not persisted to S3 as it isn't a very long running build operation in comparison to other operations, but if this keeps being flaky it might be worth looking at.

dan-v, 2018-10-12:
Latest release is now building with Ubuntu 18.04 and using debugfs. I've had better results with the limited builds I've done so far. I've also added a 30 minute timeout to the android-prepare-vendor script execution as a failsafe as well.

closed 2018-10-12 by dan-v

pgera, 2018-10-12:
Did the logging behaviour change ? Everything is going to syslog instead of cloud-init.log