dan-v/rattlesnakeos-stack #58: sed issues with prebuilt repo

pgera, 2018-10-11:
I think the sed command that adds prebuilts to PRODUCT_PACKAGES only works if there is a single LOCAL_MODULE definition in a file. I'm getting failures for for a repo similar to https://github.com/lineageos4microg/android_prebuilts_prebuiltapks. 
`sed: -e expression #1, char 73: extra characters after command`
One option may be to just add a separate manifest file for prebuilts the way it's described in the README (https://github.com/lineageos4microg/docker-lineage-cicd)

dan-v, 2018-10-11:
Ah yes, definitely not going to work in the case of multiple module definitions. The additional prebuilt stuff is definitely still in very early stages. I'll take a look at their approach used there.

dan-v, 2018-10-11:
I may just go with the manifest approach just like the custom patching currently uses (https://github.com/RattlesnakeOS/community_patches). It would keep same directory structure but manifest file would just define the product packages you want to add to the build rather than trying to guess based on contents of Android.mk file. Thoughts?

pgera, 2018-10-11:
Yes that sounds more robust. You mean a manifest file in the remote prebuilt repo , right ? I'm not too familiar with the build process, but does it cover the ones in the linked prebuilt repo in the previous comment? For eg., this Android.mk file https://github.com/lineageos4microg/android_prebuilts_prebuiltapks/blob/master/GmsCore/Android.mk defines the following two LOCAL_MODULES: privapp-permissions-com.google.android.gms.xml and GmsCore. The first one is just a permissions whitelist, and it's a dependency for GmsCore which is the second LOCAL_MODULE. Is privapp-permissions-com.google.android.gms.xml really a PRODUCT_PACKAGE ? Or does it go somewhere else ? Since it's just a file, technically it could be added as a patch as well.

dan-v, 2018-10-12:
Alright changes have been pushed in latest release to require a manifest. You can see updated example here: https://github.com/RattlesnakeOS/example_prebuilts.

closed 2018-10-12 by dan-v

pgera, 2018-10-12:
I think there is an issue with the manifest. It looks like it is ignored. It tries to build everything in the git repo even if you only specify a subset in the manifest.

The patch part succeeded:

```
2018-10-12 18:48:25: Cloning custom patches https://gitlab.com/pgera/aosp_patches to /home/ubuntu/patches
Cloning into '/home/ubuntu/patches'...
warning: redirecting to https://gitlab.com/pgera/aosp_patches.git/
2018-10-12 18:48:25: Applying patch 00002-microg-sigspoof.patch
patching file frameworks/base/core/res/AndroidManifest.xml
patching file frameworks/base/core/res/res/values/config.xml
patching file frameworks/base/core/res/res/values/strings.xml
patching file frameworks/base/services/core/java/com/android/server/pm/PackageManagerService.java
2018-10-12 18:48:25: Putting custom prebuilts from https://gitlab.com/pgera/android_prebuilts_prebuiltapks in build tree location /home/ubuntu/rattlesnake-os/packages/apps/Custom
Cloning into '/home/ubuntu/rattlesnake-os/packages/apps/Custom'...
warning: redirecting to https://gitlab.com/pgera/android_prebuilts_prebuiltapks.git/
2018-10-12 18:48:26: Adding custom PRODUCT_PACKAGES += GmsCore to /home/ubuntu/rattlesnake-os/build/make/target/product/core.mk
```

Later:
```
[412/937] including packages/apps/Custom/GmsCore/Android.mk ...
[413/937] including packages/apps/Custom/NetworkLocation/Android.mk ...
build/make/core/base_rules.mk:260: error: packages/apps/Custom/NetworkLocation: MODULE.TARGET.ETC.privapp-permissions-com.google.android.gms.xml already defined by packages/apps/Custom/GmsCore.
18:51:44 ckati failed with: exit status 1
```

NetworkLocation was not supposed to built. It wasn't in the manifest.