dan-v/rattlesnakeos-stack #60: Investigate local manifest option for advanced customization options

dan-v, 2018-10-16:
Currently customization options like --repo-patches have a remote manifest that define what patches should be included in a build. An alternative approach may be to include a local manifest that defines a list of repositories and patches to include in the build process.

Example:

```
aosp_patches:
    https://repo1
        0001-patch.patch
        0003-patch.patch
    https://repo1
        0001-patch.patch
```

dan-v, 2018-10-16:
@pgera - I've been thinking about this in the back of my head a bit. One concern I have about this approach is that it is potentially advocating injecting a bunch of code from various unknown repositories not in your control. This is dangerous because the remote code could change at any time; it could do really anything that it wanted (grab signing keys, silently inject malicious code, etc). Certainly the currently implementation of the feature could also be used to point at a repo not in your control, but it's not advocating stitching together a bunch of repositories in the same fashion. With current feature if you really want to customize to your liking - it requires you to create your own repository. Thoughts?

pgera, 2018-10-16:
Yes, that's a valid concern. Ideally, you want a common community repo of patches which are vetted, and your local manifest can pick and choose from that repo. If the community patch repo is co-maintained with the project, it's ok since you are placing the same amount of trust in the build stack to begin with. You shouldn't go beyond this repo, unless you know what you are doing.

For prebuilt apks, maintaining a repo of prebuilt apks will involve tracking upstream apks in some way and updating your local repo, which seems like duplication of work. Actually, it becomes 2 layers of duplication, as it is in the case of lineage microg apks. They need to duplicate and track the upstream apks, and you need one more indirection after that if you want to use it with rattlesnake stack. This also will have the similar security ramifications if you are just trusting a repo of prebuilt apks. That's why I suggested adding remotes in the .repo/manifest.xml if things can be built from source along with aosp. It's a better option than adding prebuilt apks. However, a lot of stuff doesn't build cleanly (like microg). So the prebuilt option is useful.

dan-v, 2018-10-20:
@pgera - just wanted to say thanks for this reply. Lots of good details in there. I like the idea of not going beyond the community repo without knowing what you are doing and also think being able to add remotes to manifest is important.

dan-v, 2018-11-03:
@pgera - I've been toying with moving all configuration options to a config file and wanted to get any feedback you may have. 

The CLI flags would still work and would override the config file values. The config (TOML right now) would look something like this:
```
chromium-version = ""
device = "marlin"
encrypted-keys = "false"
force-build = true
hosts-file = "https://raw.githubusercontent.com/StevenBlack/hosts/master/hosts"
instance-regions = "us-west-2,us-west-1,us-east-1,us-east-2"
instance-type = "c5.4xlarge"
max-price = "1.00"
name = "rattlesnakeos-mystack"
region = "us-west-2"
schedule = "rate(2 days)"
skip-price = "1.00"
ssh-key = "mykey"

[[repo-patches]]
  patches = ["00001-global-internet-permission-toggle.patch"]
  repo = "https://github.com/RattlesnakeOS/community_patches"

[[repo-patches]]
  patches = ["00002-microg-sigspoof.patch"]
  repo = "https://notabug.org/bubblethink/aosp_patches"

[[repo-prebuilts]]
  apps = ["GmsCore", "GsfProxy", "FakeStore", "com.google.android.maps.jar"]
  repo = "https://notabug.org/bubblethink/android_prebuilts_prebuiltapks"
```
There is a 'setup' subcommand that will ask basic initial questions (stack name, region, etc) and generate a config file. 

The normal stack deployment/update action would move to a subcommand like 'deploy'. It would pick up the config file automatically (or you could specify a config file as well) and you wouldn't need to pass any flags to it unless you wanted to override a value. 

Some questions for you:
1) Right now I just have excluded the CLI flags for patches and prebuilts with this setup as any method to input this data would be cumbersome at best. This means you can only use patches and prebuilts if you are using a config file. Thoughts?
2) The above configuration doesn't end up building successfully as there are multiple attempts to patch the same file. I knew this would happen, but I'm trying to think of the best way to deal with it. Any ideas?

pgera, 2018-11-05:
A few general thoughts:

- Are config errors going to be dealt with on the client ? If there are issues with the config, it should error out before doing anything on AWS. Some basic sanity checks like the files, repos or packages not existing should probably done on the client.
- Once you launch a build, the subsequent automated builds that happen after 2 weeks don't require any involvement, right ?
- CLI flags overriding config file sounds right. Ideally, the tool should also spit out a config.out file that reflects the actual config used. 
- I think excluding the advanced flags from the CLI is generally OK. People already using the feature would have to change things, but I don't think there are too many users for this feature.
- About conflicts in patches, I don't know if there is a general solution. At best, you could curate a repo of patches where they are all compatible with each other. Arbitrary patches can still conflict.
- The only other thing I can think of is to keep the config file simple and general. I think most things can be represented as key-value pairs internally. For eg., for prebuilt apps, you want to check that the same prebuilt app is not included from different repos. So <app,repo> can be the key-value pair. While parsing, if you see the same key again, you know there's a conflict.

dan-v, 2018-11-05:
@pgera - thanks so much for the feedback. This is very helpful.
* I'm not currently detecting error or duplicates on client side for this feature - but this is a great idea.
* This won't change the automated builds at all - still hands off without any involvement after running 'deploy' to update your stack with any new values.
* I am indeed spitting out the config file that is being used and settings that will be applied - and also have a configuration prompt to hit enter as well for good measure.
* Glad you think that the advanced flags could be excluded - because I couldn't think of any good way to include them :)
* Yeah, I guess the community repo can serve as the curated repo of patches, and if you want to pick and choose from that it may or may not work.

dan-v, 2018-11-07:
I pushed this feature in v9.0.12 with addition of moving to config file change mentioned before. Documented [here](https://github.com/dan-v/rattlesnakeos-stack/blob/9.0/README.md#customizations).

closed 2018-11-07 by dan-v

pgera, 2018-11-07:
What's the exact syntax for prebuilts ? This didn't get picked up:

[[repo-prebuilts]]
  repo = "https://notabug.org/bubblethink/android_prebuilts_prebuiltapks"
  apps = ["GmsCore", "GsfProxy", "FakeStore", "com.google.android.maps.jar"]

dan-v, 2018-11-07:
Ah sorry - changed syntax a bit. Check that doc page. Its custom-prebuilts now and modules instead of apps.