dan-v/rattlesnakeos-stack #63: New stable Chromium (v70) build is failing

dan-v, 2018-10-18:
Need to investigate.

pgera, 2018-10-27:
You either need to set enable_resource_whitelist_generation=false or set symbol_level>0 based on the error.

dan-v, 2018-10-27:
Indeed. Yeah I've fixed it but haven't committed change yet. Will be in next release.

dan-v, 2018-11-07:
Resolved in v9.0.12.

closed 2018-11-07 by dan-v

pgera, 2018-11-07:
Is this working for you ? In the current build for sailfish, chromium doesn't launch.

dan-v, 2018-11-07:
I don't think I have done a full rebuild of Chromium after manual fixing, so its possible there is an issue. If you could try to get logcat output that would be useful. Otherwise I'll try to track down.

pgera, 2018-11-07:
I've gone back to my old build for now. I'll see if I can get the logs for the latest release.

dan-v, 2018-11-07:
Ok thanks. I'm kicking off a rebuild of Chromium and will take a look. I'm gonna push a new release that pins back to old version until I can further debug.

dan-v, 2018-11-07:
I'm not able to reproduce the issue. I rebuilt 70.0.3538.80 and all is working as expected (chromium and webview). Is this the version you were having issues with?

pgera, 2018-11-07:
It was the same version. I wonder if this has something to do with microg as that would be the only deviation from stock. I need to investigate more. However, in the failed build, the microg packages didn't even get included due to the other parsing issue. So I think the single patch file would have been the only deviation. I'll do some builds with and without microg to confirm.

dan-v, 2018-11-07:
Let me know if you manage to track down the issue - I'm going to keep pinned to the previous v69 for the time being until we have more details because a broken webview breaks all the things.

pgera, 2018-11-08:
I did a build with v70 and microg, and it is working fine now. So I'm not sure what the reason for the previous failure was. That build was anyway half botched due to the apks not being included. There is a separate serious bug with the combination of microg, chromium and location permissions for google.com. google sites probably use google play services for location; that's why the fused location doesn't work. For eg., you won't get the gmaps site to show the location with fused provider.  It just keeps spinning. However, with microg, and only for google sites, it works for a bit, but crashes chromium and system ui. So as a workaround, location permission needs to be blocked for google.com in chromium if you want to use microg.