dan-v/rattlesnakeos-stack #64: Build Fingerprint

pgera, 2018-10-20:
It may be useful to use (perhaps optionally with a flag) the build fingerprint used by the official OEM images. This has two potential benefits: 

1. Play store (and hence yalp store) uses this to check compatibility of apps somewhere I think.
2. Safetynet CTS profile check also uses this I think.

Any thoughts on extracting this from the images somewhere near the prepare-vendor stage ?

dan-v, 2018-10-20:
Are you saying if you just match the build fingerprint name it passes the Safetynet CTS profile check? That's very interesting! 

Anyways, I'm thinking it sounds better as a patch that lives with other microg related patches, but could potentially be convinced otherwise if:
1) It had some useful property to people running stock RattlesnakeOS
2) It didn't have a negative impact on the resulting build (like reducing security in any way)

I'd definitely like to see you have a fully working microg build up and running either way. So if it doesn't meet that above criteria and it's difficult to integrate into the build process - let's chat about how we could make it easier.

pgera, 2018-10-20:
The microg build is already working. I linked the repos in the Reddit thread. For safety net, you need a separate Droid guard APK that's not a part of the repo. I'm not sure if I'll add it though since it has some closed code. However, if you do include it, it passes safety net.

dan-v, 2018-10-20:
That's awesome! 

Do you need to include the droid guard APK as well as this fix for safety net to work?

pgera, 2018-10-20:
Yes. Droid guard needs to be a priv-app. Hence the hesitance. Also the upstream droid guard build is not up to date. So that needs modifications too.

pgera, 2018-10-20:
The main thing we need is the extraction of default.prop in the script somewhere. I don't know if android-prepare-vendor does that already, but even if it doesn't, it's just a matter of doing simg2img system.img system.raw and mounting system.raw to get default.prop from the mounted image. This needs to happen earlier, and outside the aosp build process. So the aosp_patches flag is not a good fit for it. The fingerprint in build/core/Makefile is set as `BUILD_FINGERPRINT := $(PRODUCT_BRAND)/$(TARGET_PRODUCT)/$(TARGET_DEVICE):$(PLATFORM_VERSION)/$(BUILD_ID)/$(BUILD_NUMBER):$(TARGET_BUILD_VARIANT)/$(BUILD_VERSION_TAGS)`. So it's just a matter of setting the variables to the values extracted earlier.

About your questions from earlier, I don't think this would affect security since its just a string config parameter. I don't know if it affects functionality in some other way. It may be safer to throw it behind a flag.

dan-v, 2018-10-31:
@pgera - could a script that is part of patches not overwrite the build fingerprint? It would be a pain though to have to re-download the factory image just to get the build fingerprint and apply it has a part of a patch. A few options I can think of:
1) Leave around the extracted factory image (there's a flag you can pass android-prepare-vendor) so that a patch could access anything it needed like the build fingerprint.
2) Copy off the build fingerprint from factory image into a known location for build process so patch could reference it
3) Extract and use this fingerprint as part of normal build process (or optionally). I'm don't think this is the approach I want to take though - it would be my preference that it is something that is patchable that could live with other microg related patches.

pgera, 2018-10-31:
All the options sound roughly equivalent. The main thing we want to avoid is re-downloading the factory images. As long as the images are accessible, the rest can be done. Options 2 and 3 sound more convenient, but 1 is also fine.

packetup, 2018-11-02:
This would be incredibly useful for those wanting to pass SafetyNet. The string is only used to uniquely identify the build. https://developer.android.com/reference/android/os/Build#FINGERPRINT

dan-v, 2018-11-19:
I ended up with option 1 as it was the quickest fix. v9.0.16 now keeps around the extracted factory image data. You could get the fingerprint with a command like so in a custom script:
```
awk -F"=" '/ro.bootimage.build.fingerprint/{print $2}' ${BUILD_DIR}/vendor/android-prepare-vendor/${DEVICE}/$(tr '[:upper:]' '[:lower:]' <<< "${AOSP_BUILD}")/factory_imgs_data/system/default.prop
```

closed 2018-11-19 by dan-v

pgera, 2018-11-25:
I wonder if the timestamp in the fingerprint is used by the OTA updater, especially if other fields of the fingerprint haven't changed. How does the updater check for a new build ?

dan-v, 2018-11-25:
The fingerprint change _shouldn't_ affect the OTA updater, but I haven't tested it. The OTA updater app looks at your S3 release bucket for a file called `<device>-stable` (e.g. marlin-stable) that contains metadata of 'date timestamp build' (e.g. would look like: 2018.06.27.23 1530141237 OPM4.171019.021.D1). It reads this file `<device>-stable`, parses fields, and looks at timestamp. If a newer timestamp exists in this file, it will look for an OTA update file in form of <device>-ota_update-<date>.zip (e.g. marlin-ota_update-2018.06.27.23.zip) and downloads and installs that. So as long as modifying the fingerprint doesn't mess with the date or timestamp - it should still work.

pgera, 2019-06-18:
I have discovered that using a stale fingerprint is problematic. If you use the flash-all.sh (without the -w flag) to update your image, it creates issues with system components that are updated compared to the previous build. This typically happens when you update in the middle of the cycle when chromium or f-droid are updated. It appears that some intermediate step is skipped if the build fingerprint remains unchanged between builds. Also, the safety-net/droidguard stuff is broken, and I had no real use for it to begin with. So I have done away with the fingerprint patch.

The issue with yalp/aurora not fetching the right apps still exists, and we should be able to fix it in clean way rather than patching things. I think it is because of some difference in one of the makefile variables where it's something like google v/s aosp. If someone has time to investigate, that would be helpful.