dan-v/rattlesnakeos-stack #67: Support for Pixel 3 / Pixel 3 XL

dan-v, 2018-10-26:
I don't own one currently, so probably not going to make much progress at the moment. If support in https://github.com/anestisb/android-prepare-vendor shows up this would be easy, otherwise will need to spend time guessing and debugging what needs to be pulled over from factory images to make a working image for these new devices.

cernekee, 2018-11-02:
I think we have a couple of new issues to work through:

1) My android-prepare-vendor PR at https://github.com/anestisb/android-prepare-vendor/issues/136 seems to build a functioning unsigned (test-keys) image, but the process through which I came up with the file list wasn't very precise and I'm not sure how to tell if any devices/drivers are broken.  Would appreciate any hints from @anestisb or others on how to do this properly.

2) When I build stock blueline AOSP PD1A.180720.030 using the *official self-extracting vendor blobs*, I can make phone calls but audio in both directions is missing.  Same result for speakerphone and !speakerphone.  Same result using android-prepare-vendor too.  LTE data works, as do normal system-generated sounds.  Need to debug further.  Maybe somebody on XDA has already looked into this.  Nexus image works fine.

3) I haven't been able to get any blueline image processed with `sign_target_files_apks` to boot.  The device winds up in eio state (red screen).  The docs say this results from a verity failure.  I'm trying to figure out how to interpret the `avbtool info_image` output.  One possible anomaly I see is that the public key sha1 is the same between signed (release-keys) + unsigned (test-keys) images.  Also, release.sh only replaces the vbmeta signing key; is this correct?

Is it accurate to say that android-prepare-vendor is absolutely required in order to bundle a vendor partition signed with our user-defined keys?  I noticed that if I use the self-extracting vendor blobs, run `sign_target_files_apks`, then use `img_from_target_files` there is no vendor.img at all in the flashable zip.

I was hoping to see the reason for the eio / kernel panic by booting into TWRP and checking /sys/fs/pstore, but it's empty.  Possibly because I had to use the hotkey to get into recovery.

dan-v, 2018-11-02:
@cernekee - just wanted to quickly say thanks a ton for all the work you've done so far! I'll take a look at your message a bit further when I have a few more minutes.

dan-v, 2018-11-03:
@cernekee - let me try to answer some of your questions:
1) Unfortunately I don't know a great way to come up with the file list of what is required to be pulled from factory images. The only time I've done this exercise is with the Android 9.0 release, and I just built plain AOSP image and then compared it to the factory image. Looking at the official vendor blobs also can glean some information. Then it's mostly a trial and error game of what to include as far as I can tell. There may be some better way that @anestisb knows.
2) I haven't played around with the official vendor blobs at all. It sounds like your image is at least bootable, which is progress and good for debugging though. Looking at logcat do you see anything that stands out in terms of errors or stack traces that are related to audio? Look for errors at bootup time as well.
3) That's unfortunate that verified boot is not working for you at the moment. Are you generating your own keys? What options are you passing to sign_target_files_apks? Are you generating avb key and flashing it to device (e.g. fastboot flash avb_custom_key pkmd.bin)? 

It is my understanding that you can't use the self-extracting vendor blobs and sign them, so android-prepare-vendor is the best option I've found. Plus, you are better off with getting blobs from factory images as they are updated more frequently.

In terms of debugging that issue, having a userdebug build and checking out pstore is definitely your best bet as you won't get any useful output over logcat.

cernekee, 2018-11-03:
> That's unfortunate that verified boot is not working for you at the moment. Are you generating your own keys? What options are you passing to sign_target_files_apks? Are you generating avb key and flashing it to device (e.g. fastboot flash avb_custom_key pkmd.bin)?

My first attempt at this used mostly-stock rattlesnakeos scripts (which, of course, generate keys for you).  I did flash the new AVB key to the device.

One thing that jumped out at me from running `avbtool info_image --image vbmeta.img` on blueline vs. walleye:

```
Minimum libavb version:   1.0
Header Block:             256 bytes
Authentication Block:     320 bytes
Auxiliary Block:          2048 bytes
Algorithm:                SHA256_RSA2048
Rollback Index:           1536105600
Flags:                    0
Release String:           'avbtool 1.1.0'
Descriptors:
    Chain Partition descriptor:
      Partition Name:          system
      Rollback Index Location: 1
      Public key (sha1):       cdbb77177f731920bbe0a0f94f84d9038ae0617d
    Hashtree descriptor:
      Version of dm-verity:  1
      Image Size:            792514560 bytes
      Tree Offset:           792514560
      Tree Size:             6246400 bytes
      Data Block Size:       4096 bytes
      Hash Block Size:       4096 bytes
      FEC num roots:         2
      FEC offset:            798760960
      FEC size:              6316032 bytes
      Hash Algorithm:        sha1
      Partition Name:        vendor
      Salt:                  37d299192187b28471b4580da8662fe1d99a92341270fe69df970caebcee0c15
      Root Digest:           c5b35c6535c0dc752d27ea82a7b4f4827ced479c
      Flags:                 0
[...]
```

Only the blueline build has the Chain Partition descriptor.  This looks like it is related to AVB 2.0; [this commit](https://android.googlesource.com/platform/build/+/3e599ea) has a little more detail.  I see that the new BOARD_AVB_SYSTEM vars are enabled in the device/ makefiles for crosshatch/blueline but not for taimen/walleye.

The cdbb77177f73... public key shown in this vbmeta struct is worrying, as it corresponds to the test key in the AOSP repos:

```
$ avbtool extract_public_key --key external/avb/test/data/testkey_rsa2048.pem --output /tmp/testkey.bin
$ sha1sum /tmp/testkey.bin 
cdbb77177f731920bbe0a0f94f84d9038ae0617d  /tmp/testkey.bin
```

I was able to fix the eio/verity error by generating a new vbmeta struct that skips the descriptor chaining step entirely:

```
mkdir tmp
cd tmp
unzip ~/blueline/img.zip
avbtool make_vbmeta_image \
    --output vbmeta.img \
    --key ~/blueline/avb.pem \
    --algorithm SHA256_RSA2048 \
    --include_descriptors_from_image system.img \
    --include_descriptors_from_image vendor.img \
    --include_descriptors_from_image product.img \
    --include_descriptors_from_image boot.img \
    --include_descriptors_from_image dtbo.img \
    --padding_size 4096 \
    --rollback_index 1536105600
zip ~/blueline/img.zip *
fastboot update -u ~/blueline/img.zip
```

After locking the bootloader, this shows the yellow screen with the pubkey hash on it.  Success!

Short term, we might want to override BOARD_AVB_SYSTEM_KEY_PATH to "" to see if we can put all of the partition root hashes back into vbmeta.img for the time being.  We shouldn't need to use fancy chained descriptors if RattlesnakeOS ships updates to all partitions at once.  AIUI that feature is mainly intended for use cases where system.img gets revved without updating vendor.img.  I would assume that we always want the RattlesnakeOS vendor.img to track the corresponding Nexus vendor.img.

> It is my understanding that you can't use the self-extracting vendor blobs and sign them, so android-prepare-vendor is the best option I've found.

Looking at vendor/google_devices/blueline/proprietary/vendor.img from the self-extracting bundle:

 * The verity footer is present (so `avbtool info_image` reports the root digest + salt)
 * The root digest and salt don't match the Nexus image, hinting that there may be differences between the stock vendor partition and the AOSP vendor partition

I suspect we could use this image as-is if there was a vbmeta entry that specified the matching salt + root hash.  Not sure if it's the best long-term solution but it might be nice to offer people the option of using the self-extracting blobs rather than having to wait for an updated android-prepare-vendor release.

cernekee, 2018-11-03:
> Any luck tracking down the audio issue?

Nope. :-(  Pretty sure it affects the stock vendor blob too, though.

Sometimes the call terminates right away and I see this: https://pastebin.com/eZNtzFwP

Other times the call connects (and the other side rings) but there is no audio in either direction.

Tried disabling HD Voice / VoLTE / etc. but couldn't find the controls for that.

> Any other issues you've encountered?

Not yet, but I've been trying to nail down the AVB stuff rather than testing other OS features.

I'm leaning toward overriding the system partition signing key (and leaving chaining enabled) rather than trying to disable the chained vbmeta struct.  Overriding the system key is straightforward: just pass `--avb_system_key` when signing.  Disabling the feature entirely seems trickier.  I tried overriding BOARD_AVB_SYSTEM_KEY_PATH to "" through environment vars and on the make command line, but neither approach worked.  Could hack the makefile from build.sh with sed but that's more fragile.

Saw another anomaly where multiple copies of product.img show up, and conflict with each other.  I think this only happens if you're doing repeated builds in the same work directory, which wouldn't happen on a normal scripted rattlesnakeos build.

> I'd like to avoid self extracting option if its not getting timely security updates.

It might be worth spending some time comparing various vendor blobs to see how closely the self-extracting archives track the Nexus images.

FWIW I proposed a fix for the vendor signing problem here: https://android-review.googlesource.com/813295

lrvick, 2018-11-04:
This looks promising, and my Copperhead OS Pixel XL can't get updates anymore... so I finally caved and ordered a Pixel 3 XL and started some AOSP builds.

Currently trying to get latest AOSP working first per the official toolchains, then add in rattlesnakeos deviations bit by bit to see what explodes.

dan-v, 2018-11-04:
@lrvick - glad to see you again! Any extra debugging you can provide to get a fully working Pixel 3 XL image would be great. @cernekee has done an awesome job getting a bootable imagine in place and working through debugging some AVB issues on Pixel 3. There are still some issues (mainly audio) that we are hoping will be resolved in the next monthly update.

cernekee, 2018-11-06:
Phone audio is still broken on blueline PQ1A.181105.017.A1 :disappointed: 

It might help if we could get a couple more people to star the bug: https://issuetracker.google.com/issues/118921806

dan-v, 2018-11-06:
That's unfortunate 😞

I've starred the issue, and when I push a new release I'll mention experimental support for Pixel 3 with known audio issue and ask others to star that bug.

lrvick, 2018-11-12:
Currently I can generate a vanilla AOSP userdebug build for crosshatch but it has no working touchscreen making it hard to verify other issues.

Using this config:
https://github.com/hashbang/os/blob/master/configs/crosshatch.env

With this minimal local build script:
https://github.com/hashbang/os/blob/master/scripts/build.sh

Logcat from clean boot: http://termbin.com/haa3

cernekee, 2018-11-12:
Weird.  Guess you could attach a mouse via USB...

The EventHub messages might have a clue.  On your crosshatch I see:

```
11-12 02:06:30.457  1140  1271 W EventHub: Unable to disable kernel key repeat for /dev/input/event2: Function not implemented
11-12 02:06:30.457  1140  1271 I EventHub: wakeMechanism=EPOLLWAKEUP, usingClockIoctl=true
11-12 02:06:30.457  1140  1271 I EventHub: New device: id=1, fd=114, path='/dev/input/event2', name='uinput-fpc', classes=0x101, configuration='/system/usr/idc/uinput-fpc.idc', keyLayout='/system/usr/keylayout/uinput-fpc.kl', keyCharacterMap='/system/usr/keychars/Generic.kcm', builtinKeyboard=true, 
11-12 02:06:30.457  1140  1271 D EventHub: No input device configuration file found for device 'qpnp_pon'.
11-12 02:06:30.458  1140  1271 W EventHub: Unable to disable kernel key repeat for /dev/input/event0: Function not implemented
11-12 02:06:30.458  1140  1271 I EventHub: wakeMechanism=EPOLLWAKEUP, usingClockIoctl=true
11-12 02:06:30.458  1140  1271 I EventHub: New device: id=2, fd=120, path='/dev/input/event0', name='qpnp_pon', classes=0x1, configuration='', keyLayout='/system/usr/keylayout/Generic.kl', keyCharacterMap='/system/usr/keychars/Generic.kcm', builtinKeyboard=false, 
11-12 02:06:30.458  1140  1271 E EventHub: could not get driver version for /dev/input/mice, Not a typewriter
11-12 02:06:30.459  1140  1271 D EventHub: No input device configuration file found for device 'gpio-keys'.
11-12 02:06:30.462  1140  1271 W EventHub: Unable to disable kernel key repeat for /dev/input/event1: Function not implemented
11-12 02:06:30.462  1140  1271 I EventHub: wakeMechanism=EPOLLWAKEUP, usingClockIoctl=true
11-12 02:06:30.462  1140  1271 I EventHub: New device: id=3, fd=121, path='/dev/input/event1', name='gpio-keys', classes=0x1, configuration='', keyLayout='/system/usr/keylayout/Generic.kl', keyCharacterMap='/system/usr/keychars/Generic.kcm', builtinKeyboard=false, 
```

On blueline, this is what I see when I grep logcat and dmesg for `fts`:

```
11-11 19:34:36.211  1166  1285 W EventHub: Unable to disable kernel key repeat for /dev/input/event3: Function not implemented
11-11 19:34:36.211  1166  1285 I EventHub: wakeMechanism=EPOLLWAKEUP, usingClockIoctl=true
11-11 19:34:36.211  1166  1285 I EventHub: New device: id=1, fd=110, path='/dev/input/event3', name='uinput-fpc', classes=0x101, configuration='/system/usr/idc/uinput-fpc.idc', keyLayout='/system/usr/keylayout/uinput-fpc.kl', keyCharacterMap='/system/usr/keychars/Generic.kcm', builtinKeyboard=true, 
11-11 19:34:36.211  1166  1285 D EventHub: No input device configuration file found for device 'fts'.
11-11 19:34:36.213  1166  1285 W EventHub: Unable to disable kernel key repeat for /dev/input/event2: Function not implemented
11-11 19:34:36.213  1166  1285 I EventHub: wakeMechanism=EPOLLWAKEUP, usingClockIoctl=true
11-11 19:34:36.213  1166  1285 I EventHub: New device: id=2, fd=113, path='/dev/input/event2', name='fts', classes=0x15, configuration='', keyLayout='/system/usr/keylayout/Generic.kl', keyCharacterMap='/system/usr/keychars/Generic.kcm', builtinKeyboard=false, 
11-11 19:34:36.213  1166  1285 D EventHub: No input device configuration file found for device 'qpnp_pon'.
11-11 19:34:36.214  1166  1285 W EventHub: Unable to disable kernel key repeat for /dev/input/event0: Function not implemented
11-11 19:34:36.214  1166  1285 I EventHub: wakeMechanism=EPOLLWAKEUP, usingClockIoctl=true
--
11-11 19:34:36.226   818  1293 I tftp_server: pid=818 tid=1293 tftp-server : INF :[tftp_os_la.c, 63] open : [-1] [-1] [384] [0] [/vendor/rfs/msm/mpss/readwrite/ota_firewall/ruleset]
11-11 19:34:36.226   818  1293 E tftp_server: pid=818 tid=1293 tftp-server : ERR :[tftp_os_la.c, 70] open failed: [2] [No such file or directory]
11-11 19:34:36.226   818  1293 E tftp_server: pid=818 tid=1293 tftp-server : ERR :[tftp_server.c, 1621] open failed : [-2] [Unknown error -2]
11-11 19:34:36.226  1166  1285 I InputReader:   Touch device 'fts' could not query the properties of its associated display.  The device will be inoperable until the display size becomes available.
11-11 19:34:36.226   818  1293 E tftp_server: pid=818 tid=1293 tftp-server : ERR :[tftp_protocol.c, 1229] sending error-pkt. Code = 1, Msg = Err=2 String=No such file or directory
11-11 19:34:36.226  1166  1285 I InputReader: Device added: id=2, name='fts', sources=0x00001103
11-11 19:34:36.226  1166  1285 I InputReader: Device added: id=0, name='uinput-fpc', sources=0x01000111
11-11 19:34:36.229  1166  1166 I SystemServer: StartAccessibilityManagerService
11-11 19:34:36.234  1166  1179 D PinnerService: pinRangeStream: null
--
11-11 19:34:36.236   802   927 D CHRE    : @ 16.964: Got nanoapp list request from client ID 2
11-11 19:34:36.237  1166  1166 I ActivityManager: Override config changes=60007dfc {1.0 ?mcc?mnc [en_US] ldltr sw392dp w392dp h785dp 440dpi nrml long hdr port ?uimode ?night -touch -keyb/v/h -nav/h winConfig={ mBounds=Rect(0, 0 - 1080, 2160) mAppBounds=Rect(0, 0 - 1080, 2160) mWindowingMode=fullscreen mActivityType=undefined} s.2} for displayId=0
11-11 19:34:36.240  1166  1285 I InputReader: Reconfiguring input devices.  changes=0x00000004
11-11 19:34:36.240  1166  1285 I InputReader: Device reconfigured: id=2, name='fts', size 1080x2160, orientation 0, mode 1, display id 0
11-11 19:34:36.240  1166  1285 I InputReader: Reconfiguring input devices.  changes=0x00000004
[... cut ...]
Some possibilities include: missing FW blob, failing to initialize, unable to reach the I2C bus, wrong kernel build, or misconfigured device tree.  The first three would probably be indicated in your dmesg.

Toward the end of your logcat I also see some I2C errors related to NFC.  Does your NFC radio work?  Mine doesn't show these errors, and it works.

```
11-12 02:08:00.355   796  2188 E NxpTml  : _i2c_write() errno : 5
11-12 02:08:00.355   796  2188 E NxpTml  : PN54X - Error in I2C Write.....
```

lrvick, 2018-11-12:
dmesg: http://termbin.com/jow3

Looks like a -lot- of firmware is broken. Still digging in.

lrvick, 2018-11-12:
In attempting to solve this I tried building a kernel:

```
In file included from drivers/clk/qcom/mdss/mdss_pll_trace.h:116:0,
                 from drivers/clk/qcom/mdss/mdss-dsi-pll-10nm.c:24:
./include/trace/define_trace.h:88:43: fatal error: ./mdss_pll_trace.h: No such file or directory
 #include TRACE_INCLUDE(TRACE_INCLUDE_FILE)
                                           ^
compilation terminated.
scripts/Makefile.build:336: recipe for target 'drivers/clk/qcom/mdss/mdss-dsi-pll-10nm.o' failed
make[4]: *** [drivers/clk/qcom/mdss/mdss-dsi-pll-10nm.o] Error 1
scripts/Makefile.build:650: recipe for target 'drivers/clk/qcom/mdss' failed
make[3]: *** [drivers/clk/qcom/mdss] Error 2
scripts/Makefile.build:650: recipe for target 'drivers/clk/qcom' failed
make[2]: *** [drivers/clk/qcom] Error 2
scripts/Makefile.build:650: recipe for target 'drivers/clk' failed
make[1]: *** [drivers/clk] Error 2
```

With the following: https://github.com/hashbang/os/blob/master/scripts/build.sh#L70-L86

Were you able to successfully build a kernel, or are you running on the prebuilt?

dan-v, 2018-11-12:
@cernekee - that is correct.

cernekee, 2018-11-13:
> dmesg: http://termbin.com/jow3

FWIW, on blueline I see a bunch of the same errors: "No such file or directory" for stuff under /odm and /product, I2C errors, etc.  In my experience, it is pretty typical for a fully working Android device to log a large number of warnings and errors.

No mention of "fts" or "touch" in your log, so I wonder if it's using a different controller.  Maybe try flashing the Nexus image (or just the DTB) to see if anything in dmesg changes?  Will probably need some way to get root, though.

> did you see latest update on the bug you filed that they are looking for a bug report as well?

Yeah, I was hoping somebody could just try making a call on an activated phone.  They have [an internal tool](https://www.youtube.com/watch?v=2eH5sllFdQI) that flashes the latest AOSP build in about 3 minutes.

lrvick, 2018-11-13:
Update: 

This combination yeilded a bootable crosshatch where all hardware seemingly works!

https://github.com/hashbang/os/blob/master/configs/crosshatch.env

WIll be doing some more extended testing tonight, but looks promising.

Still using prebuild kernel, and the kernel branch that is -supposed- to generate the prebuilt, can't even build, which is concerning (and exactly the reason why I think we should build kernels all the time).

lrvick, 2018-11-14:
Okay yeah. Everything works, including LTE -except- for audio during a phone call. Same issue as blueline, so not surprising.

With a debug kernel this should be easy enough to spot, buuuut that would require being able to build a kernel at all, so yeah. Next step.

shareefalis, 2018-11-18:
Have you tried importing this service for incall audio?
https://github.com/phhusson/vendor_hardware_overlay/tree/master/Qualcomm/QtiAudio

shareefalis, 2018-11-18:
you will need to turn off the privapp whitelist because the QTI Audio app  require priv perm and there is no whitelist for that.
example: https://github.com/UnicornioSucio/android_device_google_crosshatch/commit/5e5ee34483517d8139d6b48ac5fb762e280c13b6