dan-v/rattlesnakeos-stack #68: Add additional documentation for encrypted keys option

dan-v, 2018-10-28:
Right now the docs aren't entirely clear about how the whole end to end process works when the encrypted keys option is enabled. Also need to in flashing guide note that avb key needs to be downloaded and decrypted for Pixel 2 and Pixel 2 XL.

cernekee, 2018-11-05:
Might also be able to use the rattlesnakeos scripts (mostly as-is) to build your images in the cloud, then re-sign them locally using a different set of keys:

```
cd $AOSP
source build/envsetup.sh
lunch aosp_blueline-user
m -j4 otatools-package
aws s3 cp s3://rattlesnakeos-$USER-release/blueline-target/blueline-target-files-2018.11.04.21.zip out/target-files-orig.zip
./build/tools/releasetools/sign_target_files_apks -o -d keys/blueline \
        --avb_vbmeta_key keys/blueline/avb.pem \
        --avb_vbmeta_algorithm SHA256_RSA2048 \
        --avb_system_key keys/blueline/avb.pem \
        --avb_system_algorithm SHA256_RSA2048 \
        out/target-files-orig.zip out/target-files-resigned.zip
./build/tools/releasetools/ota_from_target_files --block \
        -k keys/blueline/releasekey out/target-files-resigned.zip \
        out/ota_update.zip
```

So obviously you would want the Updater app on the phone to look for a different filename, rather than downloading OTA updates signed with the "throwaway" releasekey stored in the cloud.

The threat models I'd consider are:

 * If you're worried about easy passive attacks (stealing keys from S3 while the VM is not running), encrypt your keys while they are at rest and decrypt for each build.
 * If you're worried about less-trivial passive attacks (e.g. somebody taking a snapshot of your VM while it's running), build in the cloud and sign locally.
 * If you're worried about active attacks (somebody messing with the source files, toolchain, etc. to compile malware into your images), build everything locally.

You should also be able to store your keys in an HSM (SignApk takes a `-providerClass` option for this purpose) although it looks like it may require changes to the scripts.

dan-v, 2018-11-05:
Thanks @cernekee. I also had thought of that approach as a more secure alternative, but decided to meet somewhere in the middle as builds of original Pixel and Pixel XL require the kernel to be built with verity key and then I would have to build things offline and it just seemed a little messy and too manual. 

I would definitely consider an approach that supports signing locally and just supports the newer Pixel devices as I think this would support a different threat model that isn't covered by the other approaches.

dan-v, 2018-11-07:
Added [some more documentation](https://github.com/dan-v/rattlesnakeos-stack/blob/9.0/README.md#whats-the-difference-between-the-default-option-and-encrypted-signing-keys-option-and-what-one-should-i-use). Opened a new issue for building in cloud and signing locally #71.

closed 2018-11-07 by dan-v