dan-v/rattlesnakeos-stack #69: Add support for Sprint

dan-v, 2018-10-31:
Request for adding support for Sprint. Not a high priority item and I'd only consider it if the android-prepare-vendor full config provides working support for it as that would be as simple as a changing a flag when calling the script.

dan-v, 2018-10-31:
@thatchrisblack - Sorry for not answering your original question. I believe for Sprint to work you need both DMService and SprintDM included from factory image which is not in the android-prepare-vendor 'naked' config that is used by RattlesnakeOS but is in the ['full' config](https://github.com/anestisb/android-prepare-vendor/blob/master/sailfish/config.json#L559) which I'm guessing is all that is required to allow Sprint to work, but I'm not positive.

lrvick, 2018-11-15:
Making sprint work will require their nasty OMA-DM backdoor on your phone to do the proprietary tower handshakes. This binary also demands very high permissions on the phone. CopperheadOS specifically refused to support it for this reason.

dan-v, 2018-11-18:
Yeah, I was thinking about still providing an option with the full config, but going to just close this one out. It goes against the privacy/security mentality of this project.

closed 2018-11-18 by dan-v