dan-v/rattlesnakeos-stack #71: Support building in cloud and signing images locally

dan-v, 2018-11-07:
Just going to pull over comment from other issue.

"Might also be able to use the rattlesnakeos scripts (mostly as-is) to build your images in the cloud, then re-sign them locally using a different set of keys:

```
cd $AOSP
source build/envsetup.sh
lunch aosp_blueline-user
m -j4 otatools-package
aws s3 cp s3://rattlesnakeos-$USER-release/blueline-target/blueline-target-files-2018.11.04.21.zip out/target-files-orig.zip
./build/tools/releasetools/sign_target_files_apks -o -d keys/blueline \
        --avb_vbmeta_key keys/blueline/avb.pem \
        --avb_vbmeta_algorithm SHA256_RSA2048 \
        --avb_system_key keys/blueline/avb.pem \
        --avb_system_algorithm SHA256_RSA2048 \
        out/target-files-orig.zip out/target-files-resigned.zip
./build/tools/releasetools/ota_from_target_files --block \
        -k keys/blueline/releasekey out/target-files-resigned.zip \
        out/ota_update.zip
So obviously you would want the Updater app on the phone to look for a different filename, rather than downloading OTA updates signed with the "throwaway" releasekey stored in the cloud.
```

The threat models I'd consider are:

* If you're worried about easy passive attacks (stealing keys from S3 while the VM is not running), encrypt your keys while they are at rest and decrypt for each build.
* If you're worried about less-trivial passive attacks (e.g. somebody taking a snapshot of your VM while it's running), build in the cloud and sign locally.
* If you're worried about active attacks (somebody messing with the source files, toolchain, etc. to compile malware into your images), build everything locally.
You should also be able to store your keys in an HSM (SignApk takes a -providerClass option for this purpose) although it looks like it may require changes to the scripts."

sihoang, 2019-10-24:
Please correct me if I am wrong.  When I sign locally, how could I verify that the build is actually legit?
If one is worried about the AWS being compromised, he/she should not trust the build downloaded from there right?

sihoang, 2019-10-24:
nvm, sorry I missed this

> If you're worried about active attacks (somebody messing with the source files, toolchain, etc. to compile malware into your images), build everything locally.

closed 2022-03-05 by dan-v