dan-v/rattlesnakeos-stack #73: [Discussion] priv-ext for yalp

pgera, 2018-11-08:
Recent changes to app delivery in the play store makes it impossible to install split apps unless the app has android.permission.INSTALL_PACKAGES (https://github.com/yeriomin/YalpStore/issues/532#event-1888807502). I think the clean way of supporting this is by doing a priv-ext app similar to f-droid's priv-ext, but one that trusts yalp. I think this can be done as a patch by creating a copy of the f-droid priv-ext app, and changing the necessary fingerprint fields and the name of the app. So in principle, such a patch should work, but I haven't tried it yet. What do you think ?

dan-v, 2018-11-08:
Very interesting. I was not aware of split APKs so thanks for bringing this to my attention. Yeah, I think the patch approach of copying fdroid priv extention and making modifications should definitely work. If you want to give it a shot I'd be happy to have that as a patch in the community repo.

packetup, 2018-11-09:
The Aurora Store is a fork of YalpStore (https://gitlab.com/AuroraOSS/AuroraStore). It has a privileged extension (https://gitlab.com/AuroraOSS/AuroraServices) which is a fork of the F-Droid Privileged Extension.

pgera, 2018-11-09:
Interesting. Does it work ?

dan-v, 2018-11-09:
@pgera - yes I think that is a good use case for sure. I added a new issue #76 to track adding this feature.

closed 2019-09-16 by dan-v