dan-v/rattlesnakeos-stack #74: SSH config not added when encrypted keys is specified

flyinggrizzly, 2018-11-09:
### Prerequisites

* [x] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [x] I am able to reproduce this issue without any advanced customization options like --repo-patches or --repo-apks flags

### Description

Deploying a stack with `encrypted-keys = true` in the config triggers SSH key supply requests as expected, but SSH attempts time out.

Manually opening port 22 on the default security group (as explained in the readme for accessing the building instance) resolves the issue.

### Steps to Reproduce

1. Deploy a stack with encrypted keys enabled
2. Trigger a build
3. Attempt to SSH in when the email is received
4. Wait for SSH to time out

**Expected behavior:**

Either access or a warning that the key is unknown

**Actual behavior:**

Time out

### Platform

OSX

### Full Command Used for Setup

```
% ./rattlesnakeos-stack deploy --config-file file.toml
% ./rattlesnakeos-stack build start --config-file file.toml
```

dan-v, 2018-11-09:
Thanks for the report! I need to fix the issue properly at some point and not require manually opening the default security group. Prior to the encrypted keys option being added it was less critical. For now, where do you think the best place in documentation is to mention this manual procedure?

The other part this reminds me of is the the SSH keypair needs to be deployed in all possible regions where your instance may launch, which by default is in 4 regions. So I'll need to document that as well.

flyinggrizzly, 2018-11-09:
Yea, I ran into that keypair location issue too.

I think in the current docs, it's probably worth noting early on when the encrypted keys option is first mentioned, and then it might be good to move all of the encrypted key stuff into its own file--there's a lot of information throughout the readme and it can be tricky to find things without resorting to in-page search.

I'm more than happy to jump on some of this documentation stuff when my move is over too. What do you think about adding a `docs/` directory and having multiple focused doc pages in there? Might help tighten up the readme and make it easier to find things?

dan-v, 2018-11-27:
For now, in v9.0.19 I've added a disclaimer to the email notification that you might need to open up SSH in the default security group. Also, made a step in prerequisite to upload public key to all regions. And now will throw an error if it doesn't find the specified ssh key in region when doing encrypted builds. Still happy to take any documentation updates @flyinggrizzly  :)

closed 2018-11-27 by dan-v

flyinggrizzly, 2018-11-27:
Docs will come--moving has been a much bigger ordeal than I expected, and I'm on a trip to the US in a couple days (which might actually afford me some time to write some up). Worst case, I'll use some time over holiday break to open up a PR or two.

dan-v, 2018-11-27:
No worries or rush @flyinggrizzly 👍