dan-v/rattlesnakeos-stack #75: --force-build flag should be available on the build subcommand, not deploy

flyinggrizzly, 2018-11-09:
### Prerequisites

* [x] I am running the [latest version](https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [x] I am able to reproduce this issue without any advanced customization options like --repo-patches or --repo-apks flags

### Description

The `--force-build` flag feels misplaced on the `deploy` subcommand. In order to force a build, it's necessary to redeploy the stack.

### Steps to Reproduce

1. Deploy stack
2. Trigger a build
3. Try to trigger a second build
4. Redeploy stack with `--force-build`
5. Trigger new build
6. Redeploy stack without `--force-build`

**Expected behavior:** I'd expect to force a build once with `./rattlesnakeos-stack build start --config-file file.toml --force-build`

**Actual behavior:** Ability to builds appears to be a state for the stack

dan-v, 2018-11-09:
This is great feedback and I totally agree. Now that there is a build command it is definitely the right place for it. I'll need to think about the best way to fix this though. I do use the stack state force-build setting on my stacks so that I can have continuous builds running (regardless of versions changing or not). What do you think about keeping the stack state setting of force-build (or maybe rename it to some sort of "always build option") but allow the 'build start --force-build' to override it?

flyinggrizzly, 2018-11-09:
I think that makes a lot of sense--if you always want to trigger a build, it makes sense to have it applied to the deployment, but it would be nice to have it available as a one-off on the `build` command to override the stack's settings.

I think separating the names also would be good. Maybe 

`only-build-when-source-changed = true` in config and,

`--force-build` on the build command?

There's the question of how to handle the transition of config files for anyone who has it set up, but that's probably a smallish number of folks? I suppose the config key could be an alias of `force-build`...

dan-v, 2018-11-09:
Thanks - yeah I think I'll go with an approach similar to this.

As for the transition of config files - yeah I'm guessing it's a smallish number of folks who use this. I'll just default the stack setting to not forcing builds (or whatever the final option ends up being called) so that it's not mandatory to manually alter an existing config file to include this option.

dan-v, 2018-11-19:
Fixed in v9.0.16. The build subcommand now has a `--force-build` flag that forces a build as expected. I ended up with deprecating the stack setting `force-build` and replacing it with `ignore-version-checks`.

closed 2018-11-19 by dan-v