dan-v/rattlesnakeos-stack #76: Add custom option to include additional remotes and projects into manifest file

dan-v, 2018-11-09:
This would allow for easy additions of projects designed to be included into the AOSP build tree.

pgera, 2018-12-31:
I was thinking of adding wireguard's remote to the manifest.

wg's documentation states adding this:
```
<remote name="zx2c4" fetch="https://git.zx2c4.com/" />
<project remote="zx2c4" name="android_kernel_wireguard" path="kernel/wireguard" revision="master" sync-s="true" />
````
There are some differences from your example in the README (notably the sync-s field is missing in the example and the revision appears only for remote and not the project). From the code, it looks like you are parsing these fields manually. It may be easier to just take the xml file from the user and add it to the directory.

pgera, 2018-12-31:
I manually spliced in the remote and project, but that didn't really patch the kernel, presumably because marlin kernel is rebuilt. So, this needs a bit more thought for a clean solution.

pgera, 2018-12-31:
I made it work with this less than ideal patch to the build script.

```
diff --git a/templates/build_template.go b/templates/build_template.go
index e5b34d7..a58e014 100644
--- a/templates/build_template.go
+++ b/templates/build_template.go
@@ -743,6 +743,12 @@ rebuild_marlin_kernel() {
   cd "${MARLIN_KERNEL_SOURCE_DIR}"
   log "Checking out kernel commit ${kernel_commit_id}"
   git checkout ${kernel_commit_id}
+  pushd .
+  cd ..
+  git clone https://git.zx2c4.com/android_kernel_wireguard
+  cd android_kernel_wireguard
+  ./patch-kernel.sh "${MARLIN_KERNEL_SOURCE_DIR}"
+  popd
 
   # run in another shell to avoid it mucking with environment variables for normal AOSP build
   bash -c "\

```

To summarize, the two issues are:

-  The custom manifest options are a bit rigid and won't work as is. 

-  Even if you make the manifest option work, sailfish/marlin are problematic due to the kernel rebuild.

sjdrc, 2019-01-01:
zx2c4 said in #6 that the kernel module requires root at the moment in any case. According to the xda support thread, the module will still work but the android app will not be able to detect if it is installed and will still use the userspace version.

pgera, 2019-01-09:
@dan-v , any thoughts on this ? Can the kernel rebuild be avoided ? If not, a kernel-patches type knob may help (not just for wg, but any kernel patches in general), but it would only apply to marlin since that's the only one that's rebuilt.

dan-v, 2019-01-09:
@pgera - the kernel rebuild is needed for marlin/sailfish in order to include the verity key to support their older (less secure) verified boot mechanism. I have thought about a kernel patches type knob before, feel free to open a feature request for that.