dan-v/rattlesnakeos-stack #78: Add support for remote attestation

dan-v, 2018-11-12:
For newer devices that support it, could include https://github.com/AndroidHardening/AttestationServer with modification to match personal signing keys.

packetup, 2018-11-12:
The only problem I see with this is no software license has been specified. Hopefully you can convince him to select an appropriate one.

thestinger, 2018-11-13:
A permissive license will be provided once the work is funded, just as was done for https://github.com/AndroidHardening/hardened_malloc.

thestinger, 2018-11-13:
You would need to build https://github.com/AndroidHardening/Auditor and https://github.com/AndroidHardening/AttestationServer with custom app signing keys and set those as the release key in AttestationProtocol since it verifies the integrity of the app via the chain of trust provided by key attestation. You would also need to replace the verified boot fingerprints in the SampleOS entries with the correct fingerprint (most easily obtained by running a modified version of the app) along with renaming it.

thestinger, 2018-11-13:
You would need to build https://github.com/AndroidHardening/Auditor and https://github.com/AndroidHardening/AttestationServer with custom app signing keys and set those as the release key in AttestationProtocol since it verifies the integrity of the app via the chain of trust provided by key attestation. You would also need to replace the verified boot fingerprints in the SampleOS entries with the correct fingerprint (most easily obtained by running a modified version of the app) along with renaming it to the correct name at least for what gets displayed to the user.

dan-v, 2018-11-15:
@thestinger - thanks for the details. With the current status, since there is no license, what is allowed and not allowed? Would you allow this to be integrated into RattlesnakeOS - or shall I wait until this work is funded and a proper license has been provided?

thestinger, 2018-11-15:
Nothing is really allowed with the sources beyond what the implicit GitHub license allows. If an OS has official releases it can be whitelisted in the code as I used to do for CopperheadOS.

dan-v, 2018-11-15:
Unfortunately that won't work as there are no official public releases for RattlesnakeOS and each person has their own signing keys. I'll just hold off for now and revisit in the future when you have a license in place. Thanks @thestinger!

dan-v, 2019-05-18:
Feature has been added in v9.0.26. Can read more about it here: https://github.com/dan-v/rattlesnakeos-stack#remote-attestation

closed 2019-05-18 by dan-v