dan-v/rattlesnakeos-stack #79: Shell script fails

packetup, 2018-11-12:
I've tried adding your custom-boot-animation shell script (https://github.com/RattlesnakeOS/example_patch_shellscript) into my patches list but the patch fails to apply.

[Date and time]: Applying patch 90000-custom-boot-animation.sh
patch: **** Only garbage was found in the patch input.

dan-v, 2018-11-12:
Make sure you are including it in your config as "scripts" and not "patches" like shown in repo readme. Also, please include the bug template with bug reports to help debug properly.

packetup, 2018-11-12:
Oops, that was exactly the issue. I didn't realize you've now separated out patches from scripts. I've updated my config accordingly.

closed 2018-11-12 by packetup

dan-v, 2018-11-12:
No worries. Its definitely not super clearly documented.

packetup, 2018-11-12:
Perhaps you can expand on this distinction even more by adding a [[custom-scripts]]

dan-v, 2018-11-13:
That's a good idea! I like it.