dan-v/rattlesnakeos-stack #81: Add tethering support for additional carriers

dan-v, 2018-11-20:
It works fine on T-Mobile, but on other carriers it may be locked down. Disabling tethering provisioning and tether dun (dial up networking) setting appears to be the workaround, but needs to be tested.
* https://github.com/AndroidHardeningArchive/platform_frameworks_base/commit/ac8cd3ecf7d34bb609b659ea0359990134141ec7
* https://github.com/AndroidHardeningArchive/platform_build/commit/1497d078159ce923d39fbf7f392e5782f4d515e4

pgera, 2018-11-21:
This is just for documentation, but the TTL also seems to be a leakage source. If there are no side-effects, the TTL of tethered connection should be set to the same as that of regular traffic. At least some carriers use this method.

dan-v, 2018-11-27:
Pushed a change for this in v9.0.19. I'll re-open if this doesn't fix it.

closed 2018-11-27 by dan-v