dan-v/rattlesnakeos-stack #82: Email address optional

packetup, 2018-11-21:
**Is your feature request related to a problem? Please describe.**
If a user doesn't want email notifications they still have to add an email address to the config that will subsequently receive an AWS Notification.

**Describe the solution you'd like**
Change the email config option from mandatory to optional. If the email config option is left empty don't subscribe to a topic.

dan-v, 2018-11-25:
I feel like it's a bit dangerous from a cost perspective to not include notifications, which is why I decided to make it mandatory. It's also almost a requirement for encrypted key builds, as you need to SSH into the instance and provide your GPG passphrase and without an alert you won't know to do this.

dan-v, 2018-11-27:
Going to close this out, I want to keep it mandatory.

closed 2018-11-27 by dan-v