dan-v/rattlesnakeos-stack #83: build start doesn't

cmetzinnernet, 2018-11-25:
### Prerequisites

* [X ] I am running the v0.9.18(https://github.com/dan-v/rattlesnakeos-stack/releases) of rattlesnakeos-stack
* [ X] I am able to reproduce this issue without any advanced customization options like --repo-patches or --repo-apks flags

### Description

rattlesnakeos-stack build start doesn't actually start any instances/build, no debug output

### Steps to Reproduce

./rattlesnakeos-stack deploy
./rattlesnakeos-stack build start
./rattlesnakeos-stack build list

**Expected behavior:** [What you expected to happen]

List should show something happenning, and EC2 console should show some instances.

**Actual behavior:** [What actually happened]

$ ./rattlesnakeos-stack build start
INFO[0000] Using config file: /home/cmetz/.rattlesnakeos.toml 
INFO[0000] Successfully started manual build for stack <name>
$ ./rattlesnakeos-stack build list
INFO[0000] Using config file: /home/cmetz/.rattlesnakeos.toml 
INFO[0000] Looking for builds for stack <name> in the following regions: us-west-2,us-west-1,us-east-1,us-east-2 
INFO[0001] No active builds found                       

### Platform
Linux

### Full Command Used for Setup

chromium-version = ""
device = "blueline"
email = "<email address>"
encrypted-keys = false
hosts-file = ""
ignore-version-checks = false
instance-regions = "us-west-2,us-west-1,us-east-1,us-east-2"
instance-type = "c5.4xlarge"
max-price = "1.00"
name = "<name>"
region = "us-east-1"
schedule = "cron(0 0 10 * ? *)"
skip-price = "0.68"
ssh-key = "test"

### Carrier
N/A

### Email Notification output
None. I did receive and confirm the notifcation subscription from the deploy step.

### Full log
There are no log files in my S3 bucket, and only one bucket was created. It contains only one file, terraform.json

There doesn't appear to be any local logging or a local verbose/debug flag. That would be very helpful for debugging problems like this.

dan-v, 2018-11-25:
It looks like your deploy step did not complete successfully. There should be a number of additional S3 buckets that got setup after running this command. Can you try to deploy again and post output? I'm also assuming you redacted stack name and email from config posted above?

cmetzinnernet, 2018-11-25:
Yes, those two fields are redacted, they are real in the actual file.

Here's the rattlesnake-stack deploy output:
[deploy.txt](https://github.com/dan-v/rattlesnakeos-stack/files/2613643/deploy.txt)

It doesn't obviously show any problems. I end up with six S3 buckets showing in the console.

(Aside/Feature request: please consider making config auto-generate a stack name. It took me a while to figure out that it's effectively a password and must be globally unique across AWS. Those restrictions weren't sufficiently clear from the README, but it might be more sensible to have config auto-generate a sensible default, that's one less thing to go wrong)

dan-v, 2018-11-26:
It looks like the deploy command worked as expected this time, as the output shows it setting up all of the remaining AWS infrastructure. I'd kick off a build now and see how it goes.  Make sure you subscribe to the email, so you get some idea of what is going on with the build process.

Autogenerating isn't a bad idea - I'll add a feature request for it.

cmetzinnernet, 2018-11-26:
I did that right after the deploy, exactly the same as before. Said it succeeded but did nothing.

dan-v, 2018-11-26:
Are you subscribed to the email alerts? It should at least be sending a message with a success or failure. You can go to the Lambda web console and execute the function manually (that is all the build start command is doing). There really isn't anything interesting on the client side to log because its just hitting a single API to execute a Lambda function.

cmetzinnernet, 2018-11-26:
I am subscribed to the email alerts. I do not get anything. I see the function in the lambda console. I did a test start with the default json parameters, but nothing appeared to happen. If I hit the monitoring tab, I can see the graph showing invocations, and I can also see the graph showing errors steady at 1.0.

At a glance, the Python code looks good.

dan-v, 2018-11-26:
Strange. Definitely haven't seen this issue before. The current code sets 644 permissions on the generated python file before zipping it up and uploading it to Lambda. If you download the Lambda function from the console (Action->Export function->Download deployment package), and unzip it can you check the permissions that are set on lambda_spot_function.py?

```
-rw-r--r--@ 1 me  staff  7529 Nov 25 05:14 /tmp/lambda_spot_function.py
```

cmetzinnernet, 2018-11-26:
Archive:  <stack>-build-9e09c5aa-86c5-48c6-8324-901a2d86bd8e.zip
Zip file size: 2909 bytes, number of entries: 1
-rw-------  2.0 unx     7496 bX     2731 defN 18-Nov-25 18:52 lambda_spot_function.py
1 file, 7496 bytes uncompressed, 2731 bytes compressed:  63.6%

unzipping the file locally results in the same, 0600.

What's your umask?

cmetzinnernet, 2018-11-26:
If I set my local umask to 0022 (the default for many *IX distributions), the script that gets uploaded to Lambda comes back with permission 0644. So it looks like the rattlesnakeos-stack code isn't handling local umask/permissions correctly. Could you please take a look at that? Reproduce by setting your local umask to 0077.

How long should I wait between build start and expecting build list to show a result? AWS takes its sweet time firing up instances.

Running the permission-fixed lambda from the console appears to work and fire up an instance. In my log, there's a warning in the log where it fired up an instance in a zone where I didn't configure a key. Should that be an error condition instead?

dan-v, 2018-11-27:
I pushed a fix in v9.0.19 that hopefully fixes the umask issue.

It should take just a minute or two for the instance to show up. Spot instances are a bit slower because it has to first bid on a spot instance before it gets launched.

An SSH key isn't necessarily mandatory unless you are using the encrypted signing keys option. If you want to poke around or debug what's going on during a build - you will still want it though. I've updated the readme prerequisite step to include how to copy your SSH public key to all regions.

cmetzinnernet, 2018-12-01:
v9.0.19 fixes this issue. Thank you.