dan-v/rattlesnakeos-stack #85: Use different malloc implementation?

szepnapot, 2018-11-26:
Hello, first of all thanks for this project @dan-v.

I want to ask that is there any way to switch AOSP's malloc to [this](https://github.com/AndroidHardening/hardened_malloc) for example?

Cheers,
Peter

dan-v, 2018-11-26:
It should be possible in the future once this has been integrated with Bionic. There is an open issue for that here: https://github.com/AndroidHardening/hardened_malloc/issues/31. I'll leave this open as a feature request.

viktara, 2019-02-10:
https://github.com/AndroidHardening/hardened_malloc/issues/31 has been closed now, what would the next steps be to include this in the rattkesnakeos build?

dan-v, 2019-02-10:
I still wouldn't consider this stable at this point in terms of it not causing issues with other Android components. Rather than forcing this into standard build process, I'm more likely add an optional hook that could allow you to replace components in the standard build manifest.

viktara, 2019-02-12:
OK, that would be cool. I'm happy to get a second device for testing purposes.

dan-v, 2019-09-16:
Closing in favor of #129

closed 2019-09-16 by dan-v