dan-v/rattlesnakeos-stack #86: Provide a better development environment

dan-v, 2018-11-27:
Right now there isn't a great way for people to easily spin up a development environment to hack on the build script or create custom patches. I have a private image with AOSP and Chromium pre-synced that I spin up and copy over build script to more quickly iterate, but it's still manual and not an ideal workflow.

ubergeek77, 2019-07-23:
I haven't yet set up rattlesnakeos-stack, so I can't speak concretely here, but I have some ideas about how to approach this that may be helpful.

The majority of rattlesnakeos-stack's heavy lifting is done by a bash script, defined in `build_template.go`. At build time, this build script has two main components that a prospective patch/script developer's computer won't have:

- Initialized ${environment_variables}
- A mounted location containing the AOSP source tree

Again, I'm still working on getting rattlesnakeos-stack set up for myself, but assuming the AOSP source tree is stored somewhere, like S3, and doesn't have to be downloaded every time, a potential development environment could look like this:

1. The rattlesnakeos-stack binary accepts a command to set up a temporary development environment
2. A new instance on AWS is launched
3. The rattlesnakeos-stack binary mounts the already-downloaded AOSP source tree somewhere to the instance (assuming a build has been run before; I suppose it could also be downloaded for the first time during this step)
4. The rattlesnakeos-stack binary copies a new bash script to the instance, i.e. `env-setup.sh`, which sets all environment variables that the build script could possibly expect for a given set of input arguments (as in, set all environment variables assuming `DEVICE=crosshatch`, for example); it's basically a dry run of the build script setting all AOSP-build-relevant environment variables, without actually building anything.
5. The rattlesnakeos-stack binary drops into a shell, runs the above script, and then turns an interactive shell session over to the user. The user is now able to freely test commands while also being able to use the same environment variables as the build script; if the user types `echo ${DEVICE}`, the shell should print, for example, `crosshatch`. This would make developing new build commands and custom scripts much easier. If the user has to manually ssh into a new shell, `env-setup.sh` can simply be run in that shell, or even set to run upon login so that it never has to be called manually.
6. On a graceful exit, some measure is taken to ensure the AOSP source tree is reverted back to the state it was in before testing; this prevents actions taken in this new proposed development environment from potentially causing builds to fail (I think the AOSP source tree is version controlled, so this should be possible?)

Now, if I'm wrong, and the AOSP source tree isn't actually stored anywhere, then this entire idea falls apart. But it _seems_ like an approach like this would work.

As far as modifying the build script goes, I personally find it very straightforward to just modify `build_template.go` before compiling rattlesnakeos-stack. Of course, a development environment like I outlined above would make it much easier to create commands that would actually be compatible with this build script template.

dan-v, 2019-07-23:
@ubergeek77 - this is great feedback and I really like the idea you laid out a lot. All of your questions on reddit and feedback here have been detailed and really spot on, so thanks for that.

I have toyed around with having a pre-downloaded AOSP source tree that I've used for development and I have some automation that builds this AMI as well. It does unfortunately cost a decent amount of money to store that image as the AOSP source tree is quite large (100GB+). The nice thing is that AMIs can be shared publicly, so if the image was built periodically and hosted in one account it wouldn't require someone developing patches/scripts to need to build and store that image themselves. So with this approach it would just be launching a custom AMI with pre-downloaded AOSP and the rest of workflow of dropping into a shell with environment variables and such I like.

ubergeek77, 2019-07-23:
You are very welcome! I'm really happy to hear that (and here I was worrying I was asking _too_ many questions!)

Using an AMI would be a great workaround to storage costs. Whoever generously hosts the AMI would incur the cost of hosting, but the AMI wouldn't necessarily even need to be updated more frequently than once a month or so.

You wouldn't miss out on AOSP patches either, even bleeding edge changes - once the AMI is mounted, you could just `repo sync` to fetch changed files from the AOSP source tree. So even if the AMI is, say, 30 days old, you'd only be looking at downloading an extra 1-2GB (if that), rather than the whole 100GB+.

As a bonus, if there is an AMI for this in place, you could _substantially_ increase the speed in which rattlesnakeos-stack gets a build going, since you can just use the AMI rather than a full download.

I know that some of the really security conscious people may raise an eyebrow to the idea of getting the AOSP source tree from Amazon (in a manner of speaking) and a potentially unknown third party hosting the AMI, rather than googlesource, but `repo` could always be used to verify that the mounted AMI is indeed untampered with, since it's all version controlled.

dan-v, 2020-01-22:
@ubergeek77 - I played around with the AMI approach discussed above a few days ago. The results aren't as promising as I was hoping for unfortunately:
* Baseline: source tree sync from scratch took `14 minutes`.
* Syncing to a new branch with AOSP source tree pre-download in AMI:
  * With the default gp2 volume, the performance was MUCH worse than just syncing the source tree from scratch. It appears it is lazy loading the EBS volume data. I ended up just cancelling the build it was taking so long (`over 30 minutes`).
  * I changed it over to a io1 volume with max 64K iops set ($8/hour I think), and this reduced it down to `3 minutes and 20 seconds`.

So if you are willing to pay for the AMI with AOSP source tree pre-downloaded and the additional cost of a io1 volume with max iops during builds, then you can shave off about 10 minutes by having an AMI in place. I think in general, this is probably not worth the additional cost.

Another option maybe worth looking at would be zipping up the AOSP source tree and pulling that down on build and unzipping. This wouldn't require an expensive io1 volume type during the build. But it would require a place to host it that had a lot of bandwidth if it were to be shared with others.

ubergeek77, 2020-01-22:
Interesting findings. I hadn't considered EBS volume performance being a bottleneck here (I've actually never encountered a situation like this until now), so that's a shame, but it makes sense.

As an alternative, my knee-jerk reaction is to suggest a similar AMI, but with artifacts from `${BUILD_DIR}/out`. But then you run into the problems of:
- How can users trust these precompiled artifacts are untampered with?
- Does this need to be updated monthly? If so, how?
- AWS may probably not load that fast enough for the build to use it.

So... not ideal.

closed 2022-03-05 by dan-v