dan-v/rattlesnakeos-stack #88: Random mac adress

ricardopampa, 2018-12-03:
Hi, Android stock offer random mac adress, why not on rattlesnakeos ?

dan-v, 2018-12-03:
Mac randomization when scanning for networks is standard now on AOSP 9.0.

ricardopampa, 2018-12-03:
but in developer mode on stock rom have random mac adress et rattlesnakeos nothing

dan-v, 2018-12-11:
I'm not familiar with what stock Android developer settings look like, but AOSP 9.0 has scanning mac randomization enabled by default which is the more critical setting in my mind. The other setting you may be seeing in developer tools is a setting for mac randomization when you are connecting to access points as well, which can cause some issues - for example on my home network I wouldn't want this enabled as it would spam the list of clients connected to my network. This setting could be enabled with the patch functionality provided by the tool - [this value](https://github.com/aosp-mirror/platform_packages_apps_settings/blob/eae6f13389e3ec913db35079ee9e44b5abec19e1/res/values/config.xml#L46) just needs to be flipped on.

closed 2018-12-11 by dan-v

pgera, 2018-12-11:
I think the default AOSP setup will set the same mac for a particular SSID although macs may be different across SSIDs. For eg., if you connect to a wifi AP with a time limit, reconnecting won't reset the timer. Does the patch solve this use case ?

Rot127, 2019-08-08:
Here is a tiny patch which is enabling the randomize MAC option in the developer settings:
https://github.com/Rot127/rattlesnake-os-patches

The patch is working fine with my Pixel 2.
The corresponding documentation can be found here: https://source.android.com/devices/tech/connect/wifi-mac-randomization

Let me know if something is not ok with it.