dan-v/rattlesnakeos-stack #90: IAM fleet role missing iam:CreateServiceLinkedRole permission

johan-adriaans, 2018-12-06:
### Prerequisites

* [X] I am running the latest version. Built from source on 2018-12-06, 9.0 branch
* [X] I am able to reproduce this issue without any advanced customization options like --repo-patches or --repo-apks flags

### Description
After the deploy I can't start the build process because the generated spot instance request is missing the "iam:CreateServiceLinkedRole" permission.

I use the following config:
````
chromium-version = ""
device = "walleye"
email = "redacted@redacted.com"
encrypted-keys = false
hosts-file = ""
ignore-version-checks = false
instance-regions = "eu-central-1,eu-west-1,eu-west-3"
instance-type = "c4.4xlarge"
max-price = "1.00"
name = "rattlesnakeos-walleye-o7g566oxb5"
region = "eu-central-1"
schedule = "cron(0 0 10 * ? *)"
skip-price = "0.68"
ssh-key = "Yubikey"
````
Note that I changed to instance-type "c4.4xlarge" (amazon won't let me start c5.4xlarge and I wanted to test the build process) and the regions to "eu-central-1,eu-west-1,eu-west-3"

### Steps to Reproduce

1. Run the deploy wit the config above
2. Start the build manually
3. Get the following error in the spot instance request:
````
Repeated errors have occurred processing the launch specification  "c4.4xlarge, ami-0bdf93799014acdc4, Linux/UNIX, eu-central-1b while  launching spot instance".
It will not be retried for at least 13  minutes. Error message: 
com.amazonaws.services.ec2.model.AmazonEC2Exception: The provided  credentials do not have permission to create the service-linked role for  EC2 Spot Instances.
(Service: AmazonEC2; Status Code: 403; Error Code: AuthFailure.ServiceLinkedRoleCreationNotPermitted)
````

Note that the role creation was successfull during the depoy:
````
aws_iam_policy.rattlesnake_spot_fleet_policy: Creating...                                                                                                                                                                                                                                   
  arn:    "" => "<computed>"                                                                                                                                                                                                                                                                
  name:   "" => "rattlesnakeos-walleye-o7g566oxb5-spot-fleet-policy"                                                                                                                                                                                                                        
  path:   "" => "/"                                                                                                                                                                                                                                                                         
  policy: "" => "{\n\"Version\": \"2012-10-17\",\n\"Statement\": [\n\t{\n\t\t\"Effect\": \"Allow\",\n\t\t\"Action\": [\n\t\t\t\t\"ec2:DescribeImages\",\n\t\t\t\t\"ec2:DescribeSubnets\",\n\t\t\t\t\"ec2:RequestSpotInstances\",\n\t\t\t\t\"ec2:TerminateInstances\",\n\t\t\t\t\"ec2:Describ
eInstanceStatus\",\n\t\t\t\t\"iam:PassRole\"\n\t\t],\n\t\t\"Resource\": [\n\t\t\t\t\"*\"\n\t\t]\n},\n{\n\t\t\"Effect\": \"Allow\",\n\t\t\"Action\": [\n\t\t\t\t\"elasticloadbalancing:RegisterInstancesWithLoadBalancer\"\n\t\t],\n\t\t\"Resource\": [\n\t\t\t\t\"arn:aws:elasticloadbalanci
ng:*:*:loadbalancer/*\"\n\t\t]\n},\n{\n\t\t\"Effect\": \"Allow\",\n\t\t\"Action\": [\n\t\t\t\t\"elasticloadbalancing:RegisterTargets\"\n\t\t],\n\t\t\"Resource\": [\n\t\t\t\t\"*\"\n\t\t]\n}\n]\n}\n"  
````

**Expected behavior:**
Not get the error and have a working spot request

**Actual behavior:**
The error above. I fixed it by adding the "iam:CreateServiceLinkedRole" permission to the role.

### Platform
OSX

### Full Command Used for Setup
rattlesnakeos-stack deploy with above config

### Carrier
Ziggo NL

### Email Notification output
The email is a bit misleading as it did create the request, but not the instance.
````
Successfully requested a spot instance.

 Stack Name: rattlesnakeos-walleye-o7g566oxb5
 Device: walleye
 Force Build: False
 Instance Type: c4.4xlarge
 Cheapest Region: eu-central-1
 Cheapest Hourly Price: $0.252600
````

### Full log
Ill add screen captures of the data to prevent the mangle of tabular data.
[image]
[image]

dan-v, 2018-12-11:
Thanks for such a detailed issue, it's much appreciated. I'm guessing this only happens on brand new AWS accounts, as I'm not able to reproduce the issue. I'll just add CreateServiceLinkedRole to the existing 'spot-fleet-policy' in next release.

johan-adriaans, 2018-12-11:
Thanks for the awesome project by the way. It is a great way to start tinkering with android and take control over my phone. I would have created a pull request but my knowledge of go is too rusty to make a useful contribution.

commit eaa25579 (2018-12-17): * Fix issue with build list command not reading regions from config file. Closes #89.
* Add iam:CreateServiceLinkedRole to spot fleet IAM policy. Closes #90.
* Update README
closed 2018-12-17 by dan-v