dan-v/rattlesnakeos-stack #93: Preferred method to include a custom library into the build

sjdrc, 2018-12-31:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I want to include the proprietary library libjni_latinimegoogle.so into the build to enable swipe typing on the AOSP keyboard. It needs to be dropped into /system/lib64. What would be the best way to achieve this?

**Describe the solution you'd like**
Either a script similar to the example replace boot animation script should directly download the .so file and copy it to the /system/lib64 folder, or a module with its own Android.mk file should be used to include the .so file into the build.

sjdrc, 2019-01-01:
Cool, thanks for that. Where is the lib64 folder relative to the BUILD_DIR? I haven't been able to figure out where to copy the file to.

dan-v, 2019-01-02:
One way to do it in a script would be to download the library into the build tree and then insert a new PRODUCT_COPY_FILES entry into one of the make files (e.g. ${BUILD_DIR}/build/make/target/product/core.mk) that looked something like (note this is untested):

```
PRODUCT_COPY_FILES += <directory location in build tree>/libjni_latinimegoogle.so:system/lib64/libjni_latinimegoogle.so
```

dan-v, 2019-01-06:
Gonna close this out, let me know if you have any more questions.

closed 2019-01-06 by dan-v